qiboteam/qibolab #795: Zurich new relevant features

Jacfomg, 2024-02-06:
Laboneq 2.21.0
- Added the ability to sweep pulse amplitudes and frequencies in near-time for readout pulses

Laboneq 2.22.0
- OpenQASM things 
- An estimate of the total run time of a compiled experiment is now accessible through the estimated_runtime property of the CompiledExperiment object. Note that this is a best-case scenario estimate and disregards any communication overhead in the execution.
- Sweeping amplitude and intermediate frequency for integration weights in a near-time loop

Laboneq 2.23.0
- Labone release Release 24.01 @hay-k could you patch this one as well please ?

alecandido, 2024-04-23:
@hay-k do we need this issue still open?

alecandido, 2024-07-16:
I assume other `laboneq` releases happened in the meanwhile (the current latest version on PyPI is 2.33.1). This issue is definitely outdated.

closed 2024-07-16 by alecandido