qiboteam/qibolab #798: Superfluous method in `Pulse` (?)

PiergiorgioButtarini, 2024-02-12:
I was looking at the code in qibolab 0.2 and I saw something that in my opinion is superfluous (but probably it isn't and you can explain me @alecandido @hay-k). 
Now that `Pulse._if` is dropped I can't see how this method https://github.com/qiboteam/qibolab/blob/51d7cf715b0d4e0e1ab7f4815244f96d8a4a82eb/src/qibolab/pulses/pulse.py#L143 would be used. In fact this method calls https://github.com/qiboteam/qibolab/blob/51d7cf715b0d4e0e1ab7f4815244f96d8a4a82eb/src/qibolab/pulses/shape.py#L77 that in order to work needs as argument the `_if` information that is not stored neither in `Pulse` nor in `PulseShape`.
So my question is, is there a usecase for https://github.com/qiboteam/qibolab/blob/51d7cf715b0d4e0e1ab7f4815244f96d8a4a82eb/src/qibolab/pulses/pulse.py#L143 ?

hay-k, 2024-02-12:
@PiergiorgioButtarini There is a PR (https://github.com/qiboteam/qibolab/pull/786) about isolating and reimplementing these modulation related functions

alecandido, 2024-02-12:
There is also another corresponding issue (#777)

PiergiorgioButtarini, 2024-02-12:
Ops, sorry I missed that. Thanks!

closed 2024-02-12 by PiergiorgioButtarini