qiboteam/qibolab #802: Can't reproduce qibolab - Tutorial - Part: Calibration

rscircus, 2024-02-13:
Hi Team,

After the friendly exchange this afternoon, I started to write a bigger PR focusing on the Tutorials. The minor errors I could fix quickly; however, I still have some reproducibility issues. For instance, compare https://qibo.science/qibolab/stable/tutorials/calibration.html to

[image]

alecandido, 2024-02-13:
This is expected, though misleading.

The plots shown are kind of demonstrative of a real calibration. But you need to calibrate something, that could be multiple things: a qubit, a whatever resonator, or an emulator.

However, in order to do it your platform should define this object, but the tutorial is only showing dummy, since it's impossible to connect to any given instrument if you don't have it (and the CI doesn't for sure...). It would be possible with a software emulator, but that's work-in-progress.

Instead, dummy is just generating random numbers, only respecting the results interface. That's what you're observing: pure noise.

So, here the issue is just related to a missing disclaimer, essentially containing what I wrote here.
I'd not change plots nor the code (until the emulator will be available), but it should be clear that the two do not match (and I agree it's not, so it's a valid issue).

scarrazza, 2024-02-14:
Off-topic: @rscircus if at some point you or your team would like to discuss about the integration or expansion of this library to support other platforms, please feel free to contact us.