qiboteam/qibolab #830: qblox sweeping problem

biankawolo, 2024-03-07:
Dear Qibo team. I am running spinq10q on qblox. The routine is  qubit spectroscopy. My action runcard is:   
- id: QUBIT WIDE
    priority: 0
    operation: qubit_spectroscopy
    parameters:
      freq_width: 50_000_000
      freq_step: 100_000
      drive_duration: 2_000
      drive_amplitude: 0.05
      nshots: 3_000
      relaxation_time: 2000. 
I am on freshly pulled qibolab and qibocal. I get the following error:
  File "/nfs/users/bianka.woloncewicz/qibolab/src/qibolab/instruments/qblox/controller.py", line 442, in _sweep_recursion
    if any(s.parameter not in rt_sweepers for s in sweepers):
RecursionError: maximum recursion depth exceeded. 

All the parameters I' m using are physically reasonable. It looks like a sweeper problem. When I dramatically reduce the freq_width to 10^6Hz it works in the sense that I don\t get the error, but I don\t see the qubit either. Also if I reduce the freq_width BUT increase the freq_step I get the same error.  Please help. Let\s save the qubits together.

alecandido, 2024-03-07:
@biankawolo could you also add the rest of the runcard? (including the platform you're using)

Just for reproducing the error.

A possible guess would be that there is a huge flaw in the sweeper recursion, that instead of recurring for each sweeper (just frequency in a spectroscopy), it is doing it for each *sweeper value* (500 values in your original runcard).
Of course this is attenuated by sweeping fewer values, but it should disappear also by increasing the `freq_step`, so I'm not much convinced this is the case (but Python has a recursion depth default limit of 1000, so something fishy should be happening anyhow).

biankawolo, 2024-03-08:
Sure! Below the parameters.json. Use the queue qblox02_tests, becaise the queue spinq10q has problems. Sometimes it doesn t want to launch a job even you are alone there. 
[parameters.json](https://github.com/qiboteam/qibolab/files/14533736/parameters.json)
 I cannot upload the platform.py (github does not allow it) so I copy-paste:
############################ ENJOY!#####################################################
import pathlib
from qibolab.channels import Channel, ChannelMap
from qibolab.instruments.qblox.cluster_qcm_bb import QcmBb
from qibolab.instruments.qblox.cluster_qcm_rf import QcmRf
from qibolab.instruments.qblox.cluster_qrm_rf import QrmRf
from qibolab.instruments.qblox.controller import QbloxController
from qibolab.instruments.rohde_schwarz import SGS100A
from qibolab.platform import Platform
from qibolab.serialize import (
    load_instrument_settings,
    load_qubits,
    load_runcard,
    load_settings,
)

NAME = "spinq10q"
ADDRESS = "192.168.0.6"
FOLDER = pathlib.Path(__file__).parent

def create():
    """QuantWare 5q-chip controlled using qblox cluster.

    Args:
        runcard_path (str): Path to the runcard file.
    """
    runcard = load_runcard(FOLDER)
    modules = {
        "qrm_rf0": QrmRf("qrm_rf0", f"{ADDRESS}:18"),
        "qcm_rf0": QcmRf("qcm_rf0", f"{ADDRESS}:8"),
        "qcm_rf1": QcmRf("qcm_rf1", f"{ADDRESS}:10"),
        "qcm_rf2": QcmRf("qcm_rf2", f"{ADDRESS}:12"),
        "qcm_bb0": QcmBb("qcm_bb0", f"{ADDRESS}:2"),
        "qcm_bb1": QcmBb("qcm_bb1", f"{ADDRESS}:4"),
    }

    controller = QbloxController("qblox_controller", ADDRESS, modules)
    twpa_pump0 = SGS100A(name="twpa_pump0", address="192.168.0.37")

    instruments = {
        controller.name: controller,
        twpa_pump0.name: twpa_pump0,
    }
    instruments.update(modules)
    channels = ChannelMap()
    # Readout
    channels |= Channel(name="L3-20", port=modules["qrm_rf0"].ports("o1"))
    # Feedback
    channels |= Channel(name="L1-1", port=modules["qrm_rf0"].ports("i1", out=False))
    # Drive
    channels |= Channel(name="L6-1", port=modules["qcm_rf0"].ports("o1"))
    channels |= Channel(name="L6-2", port=modules["qcm_rf0"].ports("o2"))
    channels |= Channel(name="L6-3", port=modules["qcm_rf1"].ports("o1"))
    channels |= Channel(name="L6-4", port=modules["qcm_rf1"].ports("o2"))
    channels |= Channel(name="L6-5", port=modules["qcm_rf2"].ports("o1"))
    # Flux
    channels |= Channel(name="L6-39", port=modules["qcm_bb0"].ports("o1"))
    channels |= Channel(name="L6-40", port=modules["qcm_bb0"].ports("o2"))
    channels |= Channel(name="L6-41", port=modules["qcm_bb0"].ports("o3"))
    channels |= Channel(name="L6-42", port=modules["qcm_bb0"].ports("o4"))
    channels |= Channel(name="L6-43", port=modules["qcm_bb1"].ports("o1"))
    # TWPA
    channels |= Channel(name="L3-10", port=None)
    channels["L3-10"].local_oscillator = twpa_pump0

    # create qubit objects
    qubits, couplers, pairs = load_qubits(runcard)

    for q in range(5):
        qubits[q].readout = channels["L3-20"]
        qubits[q].feedback = channels["L1-1"]
        qubits[q].twpa = channels["L3-10"]
        qubits[q].drive = channels[f"L6-{q+1}"]
        qubits[q].flux = channels[f"L6-{39+q}"]
        channels[f"L6-{39+q}"].qubit = qubits[q]
        qubits[q].flux.max_bias = 2.5

    settings = load_settings(runcard)
    instruments = load_instrument_settings(runcard, instruments)

    return Platform(
        str(FOLDER), qubits, pairs, instruments, settings, resonator_type="2D"
    )

biankawolo, 2024-03-08:
Sorry the parameters.json file is here: 
[parameters.json](https://github.com/qiboteam/qibolab/files/14534107/parameters.json)

hay-k, 2024-03-08:
@biankawolo thanks, I was able to reproduce the issue easily with all the details provided. It does indeed look like an issue in the driver. I will try to put up a fix soon.

hay-k, 2024-03-08:
While I am working on a fix, if you want to run experiments without sacrificing the freq weep width, one workaround would be to decrease the number of shots. If you use numbers such that `num_shots * freq_width / freq_step` is smaller than `2**17`, the error can be avoided.

closed 2024-03-18 by hay-k