qiboteam/qibolab #840: error dumping zi experiment log

DavidSarlle, 2024-03-18:
@GabrielePalazzo @hay-k when trying to save the experiment log using ZI driver (qibolab main) the following error is raised:

File "/nfs/users/david.fuentes/qibolab/src/qibolab/instruments/zhinst.py", line 1210, in sweep
    self.run_exp()
  File "/nfs/users/david.fuentes/qibolab/src/qibolab/instruments/zhinst.py", line 543, in run_exp
    self.results = self.session.run(self.exp)
  File "/nfs/users/david.fuentes/iqm5q/lib/python3.10/site-packages/laboneq/dsl/experiment/experiment.py", line 1239, in save
    Serializer.to_json_file(self, filename)
  File "/nfs/users/david.fuentes/iqm5q/lib/python3.10/site-packages/laboneq/dsl/serialization/serializer.py", line 123, in to_json_file
    json_string = Serializer.to_json(serializable_object)
  File "/nfs/users/david.fuentes/iqm5q/lib/python3.10/site-packages/laboneq/dsl/serialization/serializer.py", line 93, in to_json
    json_struct = serialize_to_dict_with_ref(
  File "/nfs/users/david.fuentes/iqm5q/lib/python3.10/site-packages/laboneq/core/serialization/simple_serialization.py", line 558, in serialize_to_dict_with_ref
    raise ex
  File "/nfs/users/david.fuentes/iqm5q/lib/python3.10/site-packages/laboneq/core/serialization/simple_serialization.py", line 538, in serialize_to_dict_with_ref
    root_object = serialize_to_dict_with_entities(
  File "/nfs/users/david.fuentes/iqm5q/lib/python3.10/site-packages/laboneq/core/serialization/simple_serialization.py", line 490, in serialize_to_dict_with_entities
    ] = serialize_to_dict_with_entities(
  File "/nfs/users/david.fuentes/iqm5q/lib/python3.10/site-packages/laboneq/core/serialization/simple_serialization.py", line 490, in serialize_to_dict_with_entities
    ] = serialize_to_dict_with_entities(
  File "/nfs/users/david.fuentes/iqm5q/lib/python3.10/site-packages/laboneq/core/serialization/simple_serialization.py", line 490, in serialize_to_dict_with_entities
    ] = serialize_to_dict_with_entities(
  File "/nfs/users/david.fuentes/iqm5q/lib/python3.10/site-packages/laboneq/core/serialization/simple_serialization.py", line 501, in serialize_to_dict_with_entities
    raise RuntimeError(
RuntimeError: uid not unique: item LinearSweepParameter(uid='frequency', start=-247826000, stop=50174000, count=150, axis_name=None) has same uid as previously encountered item {'__type': 'LinearSweepParameter', 'uid': 'frequency', 'start': -247826000, 'stop': 50174000, 'count': 150, 'axis_name': None, '__id': 133869641461968}

I am traying to save the log using the method commented on line 542:

[image]

alecandido, 2024-03-19:
Hi @DavidSarlle, thanks for the report. I'm not sure what is happening, but it seems like there is more than one swept parameter named `frequency`.

Though, in principle, that's perfectly possible (you could sweep the frequency of multiple pulses), and as such that UID is rather bad, and definitely not unique, it seems quite funny.

Could you provide also the Qibocal runcard, or (if you're not using Qibocal) the pulse sequence you're trying to sweep?

DavidSarlle, 2024-03-19:
Thanks @alecandido for checking the error. The runcard that I am trying is the following one (available in the qibocal PR https://github.com/qiboteam/qibocal/pull/691):

platform: iqm5q 
qubits: [0]
format: csv

actions:

  - id: qubit flux dependence amplitude
    priority: 0
    operation: qubit_flux
    parameters:
      freq_width: 300_000_000
      freq_step: 2_000_000
      flux_amplitude_start: -1.0
      flux_amplitude_end: 1.0
      flux_amplitude_step:  0.03
      # drive_amplitude: 0.5
      nshots: 1024
      relaxation_time: 2000
      transition: "01"
      
The action runs without errors if the report dump is not enabled. Just to let you know. If you try to run any other routine (like resonator spec) the same error is raised:

[image]

The point is that I am now characterizing the couplers of iqm5q and the flux dependence showed for all the qubits sweeping the bias is minimal (but there is a shift) after changing the way of sweeping the flux in https://github.com/qiboteam/qibocal/pull/691

But if I try sweeping the amplitude, there is no shift in the drive freq. This behavior is the same in all the qubits:

[image]

So, we are checking the flux lines in parallel, but I want to check the pulse sequence sent to the instruments and for that, we need as much information as we can get from the logs.

Also, zi html report got from the controller, does not show the qf flux pulses, but I am not sure if it should. I remember this plot showing qf pulses before, but I have no reports saved like this:

[image]

DavidSarlle, 2024-03-19:
@alecandido I have also tried with qibolab and qibocal main, and the report dump also crashes running a resonator spectroscopy for example:

[image]

alecandido, 2024-03-20:
Sorry @DavidSarlle if I didn't reply before. We're about to merge @hay-k PR, so I'd just wait for it (and in the, short, meanwhile, you could even use his branch, if urgently needed).

DavidSarlle, 2024-03-20:
I have seen the Pr @alecandido . I have a job right know in the queue to test the @hay-k PR and see the logs. Thanks! But we still have the problem with the flux. Maybe we could keep open this issue until we are sure that is not a driver problem or routine problem ([WIP] https://github.com/qiboteam/qibocal/pull/691#issuecomment-2009056449)

DavidSarlle, 2024-03-20:
@hay-k now, using your PR #797 the error is not raised, but the neither the html report nor the saved_exp log is generated. Or at least, I can not find the logs in my workspace...

hay-k, 2024-03-20:
@DavidSarlle neither main, nor my branch save the experiment or the report out of the box. You do it by adding some lines in the code. Same lines can be added regardless of the branch.

DavidSarlle, 2024-03-20:
@hay-k I know that I have to add these lines into your branch to save the files. But adding them, as I do in main, the reports are not saved (neither html nor the saved_exp)

hay-k, 2024-03-21:
> @hay-k I know that I have to add these lines into your branch to save the files. But adding them, as I do in main, the reports are not saved (neither html nor the saved_exp)

I believe this was resolved through offline discussion.

Regarding the fact that the experiment results don't show any flux dependence - I think there are two things:

1. The current version of the experiment does not add the flux pulses to the pulse sequence, hence the sweep has no effect basically. Ongoing discussion - https://github.com/qiboteam/qibocal/pull/691#discussion_r1531661327
2. Even when they are added to the sequence, I don't think the experiment will produce expected results, due to the way the ZI driver is designed currently. It will need some changes, or a hotfix. I can decide what to do based on urgency of this experiment working from your side.

hay-k, 2024-03-26:
1. The issue about dumping the ZI experiment was automatically solved with https://github.com/qiboteam/qibolab/pull/797
2. The issue about the experiment not showing flux dependence when flux pulse amplitude is used instead of bias itself is resolved via https://github.com/qiboteam/qibolab/pull/848

closed 2024-03-26 by hay-k