qiboteam/qibolab #860: Several Errors - Comparison with 15/12/2023

aorgazf, 2024-03-31:
Dear @hay-k , @andrea-pasquale 
As explained on Friday, I've run calibration of spin10q q1 using old versions of qibocal and qibolab, dated 15/12/2023:
qibocal: [alvaro/latest_20231215](https://github.com/qiboteam/qibocal/tree/alvaro/latest_20231215)
qibolab: [alvaro/latest_20231215](https://github.com/qiboteam/qibolab/tree/alvaro/latest_20231215)
qibolab_platforms_qrc: [alvaro/latest_20231215](https://github.com/qiboteam/qibolab_platforms_qrc/tree/alvaro/latest)

And then tried to run the same experiments using the main branches in both repositories.
Several of the key characterisation routines in main no longer work:
[image]
Attached are all reports and debugging information generated.
[comparison_with_20231215.zip](https://github.com/qiboteam/qibolab/files/14814208/comparison_with_20231215.zip)
Thanks for looking into this.

hay-k, 2024-04-01:
Hi @aorgazf, thanks for making the effort to report the issues. With @andrea-pasquale we skimmed through the reports with the aim of identifying concrete action points. It is a bit hard to navigate through the data since there is no one-to-one mapping for old vs new comparison, hence it is not always clear what is exactly the thing that is 'no longer working'. Further elaboration would be appreciated. Below are some quick comments from our side:

1. **20240329_spinq10q_02_punchout_att**: Not sure if there is an issue here. Everything looks good.
2. **20240329_spinq10q_01_resonator_hp_fails**: This should already be fixed via https://github.com/qiboteam/qibolab/pull/857
3. **20240329_spinq10q_06_qubit_50shots_fails**: Not sure if we understood what failed here. The acquisition was completed without errors. The data looks like just noise, so I wouldn't expect much from the fit. And I believe the fit behaved the same before as well, so there is no new behavior. We could perhaps consider improving the fit so that it refuses to fit complete noise in general, but this should be a separate discussion.
4. **20240329_spinq10q_07_rabi_amp_5kshots**: Not sure if we understood what is the problem here. For qubit 1 the data is nice and the fit is nice. For the rest - same as above.
5. **20240329_spinq10q_07_rabi_amp_50shots**: Same as above.
6. **20240329_spinq10q_03_resonator_flux_1000shots_fails**: This issue is already separately reported in https://github.com/qiboteam/qibolab/issues/845. Let's follow the progress there.
7. **20240329_spinq10q_04_resonator_50shots**: We can see that for some qubits the fit didn't work. We can explore possibilities to improve the fit in general. However, I have to note that the fitting logic has not changed recently, and in the report generated by older version of qibocal (where all fits worked) the sweep range is narrower. This makes the loretzian fit easier. So the comparison is not entirely apples to apples. Using narrow sweep ranges should mitigate the issue most of the time, however if you have any suggestion on how to evolve the fit beyond basic lorentzian, please feel free to open a separate feature request.
8. **20240329_spinq10q_09_t2_sequences_takes_60xlonger**: This one errored out, so I am not sure compared to what it took 60x longer. Nevertheless, the error itself is already separately reported in https://github.com/qiboteam/qibolab/issues/836. Let's follow the progress there.
9. **20240329_spinq10q_03_resonator_flux_50shots**: Fitting issue. Although, looking at the report from the older version of software, I can see the fit was not correct there either. So, not exactly a new behavior, but anyways the fit should certainly be improved.
10. **20240329_spinq10q_05_qubit_flux_50shots**: Fitting issue, however the data is also extremely noisy, plus the fitting was even worse in the older version. So again, no broken functionality compared to the previous version, but the fit could perhaps be improved.
11. **20240329_spinq10q_05_qubit_flux_1000shots_fails**: This is already separately reported in https://github.com/qiboteam/qibolab/issues/845. Let's follow the progress there.
12. **20240329_spinq10q_04_resonator_1000shots_fails**: Should already be fixed via https://github.com/qiboteam/qibolab/pull/857
13. **20240329_spinq10q_06_qubit_5kshots_fails**: Should already be fixed via https://github.com/qiboteam/qibolab/pull/857
14. **20240329_spinq10q_10_ramsey_signal_fails**: Not sure if we understood what is the issue here.

In summary, I think only 2. 6. 8. 11. 12. 13. are issues that can block the characterization team, hence need asap attention. All of them already have their own dedicated issues, so I will not create more. Some of them are already fixed, and the rest are in active progress, so I think everything is already happening as optimal as it gets! Let's follow the progress on the dedicated issues, and please elaborate more on points that you think we misunderstood/misinterpreted.

andrea-pasquale, 2024-04-02:
> * **20240329_spinq10q_03_resonator_flux_50shots**: Fitting issue. Although, looking at the report from the older version of software, I can see the fit was not correct there either. So, not exactly a new behavior, but anyways the fit should certainly be improved.
> 
> * **20240329_spinq10q_05_qubit_flux_50shots**: Fitting issue, however the data is also extremely noisy, plus the fitting was even worse in the older version. So again, no broken functionality compared to the previous version, but the fit could perhaps be improved.

I'm having a look at these two points in https://github.com/qiboteam/qibocal/issues/782

DavidSarlle, 2024-05-15:
Probably related with this @hay-k @andrea-pasquale. When I run in iqm5q chip the classifier, using main in qibocal and qibolab:

[image]

While using the same parameters with an old branch:

qibocal: alvaro/latest_20231215
qibolab: alvaro/latest_20231215

adapting the paramters.json to the old .yml pararmeers file:

[image]

It is the same for all qubits.

alecandido, 2025-10-20:
The problem only regarded 0.1, which by now has been discontinued.