qiboteam/qibolab #875: `make`-ing the docs generates a set of errors around bluefors and instruments

rscircus, 2024-04-21:
# Env

- macOS 14
- qibolab - commit: https://github.com/qiboteam/qibolab/commit/a514441f7c020b190c7871df90d6892392d0b4da

# Want

- `make` in `doc` folder should generate docs.

# Actual behavior:

```
Running Sphinx v6.2.1
loading pickled environment... done
File /Users/r/Repos/eth/qibo/qibolab/doc/source/api-reference/qibolab.rst already exists, skipping.
File /Users/r/Repos/eth/qibo/qibolab/doc/source/api-reference/qibolab.compilers.rst already exists, skipping.
File /Users/r/Repos/eth/qibo/qibolab/doc/source/api-reference/qibolab.dummy.rst already exists, skipping.
File /Users/r/Repos/eth/qibo/qibolab/doc/source/api-reference/qibolab.instruments.rst already exists, skipping.
File /Users/r/Repos/eth/qibo/qibolab/doc/source/api-reference/qibolab.instruments.qblox.rst already exists, skipping.
File /Users/r/Repos/eth/qibo/qibolab/doc/source/api-reference/qibolab.instruments.qm.rst already exists, skipping.
File /Users/r/Repos/eth/qibo/qibolab/doc/source/api-reference/qibolab.instruments.rfsoc.rst already exists, skipping.
File /Users/r/Repos/eth/qibo/qibolab/doc/source/api-reference/qibolab.instruments.zhinst.rst already exists, skipping.
building [mo]: targets for 0 po files that are out of date
writing output... 
building [html]: targets for 0 source files that are out of date
updating environment: 0 added, 1 changed, 0 removed
reading sources... [100%] api-reference/qibolab.instruments

/Users/r/Repos/eth/qibo/qibolab/src/qibolab/instruments/bluefors.py:docstring of qibolab.instruments.bluefors.TemperatureController:6: ERROR: Unexpected indentation.
/Users/r/Repos/eth/qibo/qibolab/src/qibolab/instruments/bluefors.py:docstring of qibolab.instruments.bluefors.TemperatureController:3: WARNING: Inline literal start-string without end-string.
/Users/r/Repos/eth/qibo/qibolab/src/qibolab/instruments/bluefors.py:docstring of qibolab.instruments.bluefors.TemperatureController:10: ERROR: Unexpected indentation.
/Users/r/Repos/eth/qibo/qibolab/src/qibolab/instruments/bluefors.py:docstring of qibolab.instruments.bluefors.TemperatureController:11: WARNING: Block quote ends without a blank line; unexpected unindent.
/Users/r/Repos/eth/qibo/qibolab/src/qibolab/instruments/bluefors.py:docstring of qibolab.instruments.bluefors.TemperatureController:11: WARNING: Inline literal start-string without end-string.
/Users/r/Repos/eth/qibo/qibolab/src/qibolab/instruments/bluefors.py:docstring of qibolab.instruments.bluefors.TemperatureController:11: WARNING: Inline interpreted text or phrase reference start-string without end-string.
/Users/r/Repos/eth/qibo/qibolab/src/qibolab/instruments/bluefors.py:docstring of qibolab.instruments.bluefors.TemperatureController.get_data:5: WARNING: Definition list ends without a blank line; unexpected unindent.
/Users/r/Repos/eth/qibo/qibolab/src/qibolab/instruments/bluefors.py:docstring of qibolab.instruments.bluefors.TemperatureController.get_data:10: WARNING: Block quote ends without a blank line; unexpected unindent.
WARNING: autodoc: failed to import module 'icarusq' from module 'qibolab.instruments'; the following exception was raised:
No module named 'icarusq_rfsoc_driver'
WARNING: autodoc: failed to import module 'icarusqfpga' from module 'qibolab.instruments'; the following exception was raised:
No module named 'icarusq_rfsoc_driver'
looking for now-outdated files... none found
pickling environment... done
checking consistency... done
preparing documents... done
writing output... [ 33%] api-reference/qibolab
writing output... [ 66%] api-reference/qibolab.instruments
writing output... [100%] index
```

rscircus, 2024-07-02:
Still seeing some:

```
> poe docs
Poe => make -C doc html
Running Sphinx v6.2.1
making output directory... done
loading intersphinx inventory from https://docs.python.org/3/objects.inv...
Creating file /Users/r/Repos/eth/qibo/qibolab/doc/source/api-reference/qibolab.rst.
Creating file /Users/r/Repos/eth/qibo/qibolab/doc/source/api-reference/qibolab.compilers.rst.
Creating file /Users/r/Repos/eth/qibo/qibolab/doc/source/api-reference/qibolab.dummy.rst.
Creating file /Users/r/Repos/eth/qibo/qibolab/doc/source/api-reference/qibolab.instruments.rst.
Creating file /Users/r/Repos/eth/qibo/qibolab/doc/source/api-reference/qibolab.instruments.qblox.rst.
Creating file /Users/r/Repos/eth/qibo/qibolab/doc/source/api-reference/qibolab.instruments.qm.rst.
Creating file /Users/r/Repos/eth/qibo/qibolab/doc/source/api-reference/qibolab.instruments.rfsoc.rst.
Creating file /Users/r/Repos/eth/qibo/qibolab/doc/source/api-reference/qibolab.instruments.zhinst.rst.
Creating file /Users/r/Repos/eth/qibo/qibolab/doc/source/api-reference/qibolab.platform.rst.
building [mo]: targets for 0 po files that are out of date
writing output...
building [html]: targets for 14 source files that are out of date
updating environment: [new config] 23 added, 0 changed, 0 removed
reading sources... [100%] tutorials/pulses
/Users/r/Repos/eth/qibo/qibolab/src/qibolab/pulses.py:docstring of qibolab.pulses.PulseShape.eval:3: ERROR: Unknown directive type "todo".

.. todo::

    To be replaced by proper serialization.
/Users/r/Repos/eth/qibo/qibolab/src/qibolab/unrolling.py:docstring of qibolab.unrolling.batch:3: ERROR: Unknown interpreted text role "cls".
/Users/r/Repos/eth/qibo/qibolab/src/qibolab/instruments/bluefors.py:docstring of qibolab.instruments.bluefors.TemperatureController:6: ERROR: Unexpected indentation.
/Users/r/Repos/eth/qibo/qibolab/src/qibolab/instruments/bluefors.py:docstring of qibolab.instruments.bluefors.TemperatureController:3: WARNING: Inline literal start-string without end-string.
/Users/r/Repos/eth/qibo/qibolab/src/qibolab/instruments/bluefors.py:docstring of qibolab.instruments.bluefors.TemperatureController:10: ERROR: Unexpected indentation.
/Users/r/Repos/eth/qibo/qibolab/src/qibolab/instruments/bluefors.py:docstring of qibolab.instruments.bluefors.TemperatureController:11: WARNING: Block quote ends without a blank line; unexpected unindent.
/Users/r/Repos/eth/qibo/qibolab/src/qibolab/instruments/bluefors.py:docstring of qibolab.instruments.bluefors.TemperatureController:11: WARNING: Inline literal start-string without end-string.
/Users/r/Repos/eth/qibo/qibolab/src/qibolab/instruments/bluefors.py:docstring of qibolab.instruments.bluefors.TemperatureController:11: WARNING: Inline interpreted text or phrase reference start-string without end-string.
/Users/r/Repos/eth/qibo/qibolab/src/qibolab/instruments/bluefors.py:docstring of qibolab.instruments.bluefors.TemperatureController.get_data:5: WARNING: Definition list ends without a blank line; unexpected unindent.
/Users/r/Repos/eth/qibo/qibolab/src/qibolab/instruments/bluefors.py:docstring of qibolab.instruments.bluefors.TemperatureController.get_data:10: WARNING: Block quote ends without a blank line; unexpected unindent.
WARNING: autodoc: failed to import module 'icarusq' from module 'qibolab.instruments'; the following exception was raised:
No module named 'icarusq_rfsoc_driver'
WARNING: autodoc: failed to import module 'icarusqfpga' from module 'qibolab.instruments'; the following exception was raised:
No module named 'icarusq_rfsoc_driver'
/Users/r/Repos/eth/qibo/qibolab/src/qibolab/instruments/qblox/module.py:docstring of qibolab.instruments.qblox.module.ClusterModule.ports:15: ERROR: Unexpected indentation.
/Users/r/Repos/eth/qibo/qibolab/src/qibolab/instruments/qblox/module.py:docstring of qibolab.instruments.qblox.module.ClusterModule.ports:17: WARNING: Block quote ends without a blank line; unexpected unindent.
/Users/r/Repos/eth/qibo/qibolab/src/qibolab/platform/platform.py:docstring of qibolab.platform.platform.Platform:1: WARNING: duplicate object description of qibolab.platform.platform.Platform, other instance in api-reference/qibolab, use :noindex: for one of them
/Users/r/Repos/eth/qibo/qibolab/doc/source/index.rst:19: WARNING: Enumerated list ends without a blank line; unexpected unindent.
/Users/r/Repos/eth/qibo/qibolab/doc/source/tutorials/lab.rst:42: WARNING: Include file '/Users/r/Repos/eth/qibo/qibolab/doc/source/tutorials/includes/lab1.py' not found or reading it failed
/Users/r/Repos/eth/qibo/qibolab/doc/source/tutorials/lab.rst:50: WARNING: Include file '/Users/r/Repos/eth/qibo/qibolab/doc/source/tutorials/includes/lab2.py' not found or reading it failed
looking for now-outdated files... none found
pickling environment... done
checking consistency... done
preparing documents... done
writing output... [100%] tutorials/pulses
/Users/r/Repos/eth/qibo/qibolab/doc/source/index.rst:19: WARNING: undefined label: 'tutorials_emulator'
generating indices... genindex py-modindex done
highlighting module code... [100%] qibolab.unrolling
writing additional pages... search done
copying images... [100%] tutorials/rotation_dark.svg
copying static files... done
copying extra files... done
dumping search index in English (code: en)... done
dumping object inventory... done
build succeeded, 19 warnings.

The HTML pages are in build/html.
~/Repos/eth/qibo/qibolab on doc/executable_tutorials !2 ....................................... took 27s py qibolab-py3.11 at 05:52:33
```

alecandido, 2026-01-28:
As the other documentations-related, also this is outdated. Sorry we never took the time to work on it

closed 2026-01-28 by alecandido