qiboteam/qibolab #883: Compiler/transpiler issue with `dummy`

andrea-pasquale, 2024-04-24:
The following code:
from qibo import set_backend, Circuit, gates

set_backend("qibolab", platform="dummy")

circuit = Circuit(1)
circuit.add(gates.X(0))
circuit.add(gates.M(0))

circuit()
```
generates the error `KeyError: "Compiler rule not available for <class 'qibo.gates.gates.X'>."`
Is this expected? I was testing some protocols in qibocal and I found the bug.

Edoardo-Pedicillo, 2024-04-24:
I think it is expected
https://github.com/qiboteam/qibolab/blob/a514441f7c020b190c7871df90d6892392d0b4da/src/qibolab/backends.py#L47-L49
( the default value of `self.transpiler` is `None`), maybe we could have something different as default transpiler, for example one with only an unroller, for sure the documentation should be updated.

alecandido, 2024-04-24:
Yes, since we moved the transpiler to Qibo, the circuits are already expected to be expressed as native gates in Qibolab, and they are just compiled to pulses.

I believe we already discussed, and for sure it should be documented (@stavros11, is there anything written anywhere?), but the idea is to avoid anything that is not strictly related to pulses, since it would duplicate a Qibo function (and circuits should be fully handled by Qibo, since you could reuse these features even outside Qibolab).

andrea-pasquale, 2024-04-24:
> Yes, since we moved the transpiler to Qibo, the circuits are already expected to be expressed as native gates in Qibolab, and they are just compiled to pulses.

I understand, however it will be useful for qibocal to have a `DummyBackend` with a basic transpiler. Of course since qibocal mainly relies on the dummy platform it would be convenient to have this feature directly on the dummy platform.

alecandido, 2024-04-25:
That could be created that in Qibocal by depending on Qibo as well. Or even in a `qibolab_platforms_qrc`-like package (eventually it could go together the emulator backends).

PR #884 (2024-05-10): Remove transpilation hook
Closes #883
closed 2024-05-10 by alecandido