qiboteam/qibolab #899: Replace all results objects in 0.2 with arrays

andrea-pasquale, 2024-05-08:
_Originally posted by @alecandido in https://github.com/qiboteam/qibolab/pull/896#pullrequestreview-2045976529_

alecandido, 2024-07-19:
The dimension of the array will be:
```py
(shots, sweeper1, sweeper2, ..., sweeperN)  # discriminated
(shots, sweeper1, sweeper2, ..., sweeperN, 2)  # IQ
(shots, sweeper1, sweeper2, ..., sweeperN, times)  # raw waveforms
```
without `shots` if an average is taken.

A function to yield the shape given `ExecutionParametrs, list[Sweeper]` will be provided (used for testing, but also exposed to the user, documenting the meaning).

alecandido, 2024-07-23:
@stavros11 @andrea-pasquale: I was considering having a fixed amount of dimensions, other than the sweeper ones. I.e. replacing "missing" dimensions with `1`s.

This operation could be automatically undone with [`np.squeeze`](https://numpy.org/doc/stable/reference/generated/numpy.squeeze.html).

The idea behind would be to have a more predictable shape, and to keep the same rank (as much as possible).
On the other side, dimensions are useful: for the same reason for which we have a variable shape length because of sweepers, it is better to have a `(143, 2)` shape for a raw waveform than a `(286,)` one (is it raw major? column major?), and better no dimension at all if discriminated.

In principle, this may complicate the handling of different execution parameters (acquisition type and averaging mode). In practice, I expect that they have to be handled separately in any case...

PR #940 (2024-08-06): Replace results with bare arrays
Resolves #899 ~~and #809~~

- [ ] test new features
- [ ] fix tests