qiboteam/qibolab #904: ERROR - Unsupported <class 'openqasm3.ast.QuantumBarrier'> statement

dgks0n, 2024-05-18:
Hi there, I tried to run below QASM script but it does not work:
```
OPENQASM 2.0;
include "qelib1.inc";
qreg qr[10];
creg cr[9];
h qr[0];
h qr[1];
h qr[2];
h qr[3];
h qr[4];
h qr[5];
h qr[6];
h qr[7];
h qr[8];
x qr[9];
h qr[9];
barrier qr[0],qr[1],qr[2],qr[3],qr[4],qr[5],qr[6],qr[7],qr[8],qr[9];
cx qr[0],qr[9];
cx qr[1],qr[9];
cx qr[2],qr[9];
cx qr[3],qr[9];
cx qr[4],qr[9];
cx qr[5],qr[9];
cx qr[6],qr[9];
cx qr[7],qr[9];
cx qr[8],qr[9];
barrier qr[0],qr[1],qr[2],qr[3],qr[4],qr[5],qr[6],qr[7],qr[8],qr[9];
h qr[0];
h qr[1];
h qr[2];
h qr[3];
h qr[4];
h qr[5];
h qr[6];
h qr[7];
h qr[8];
measure qr[0] -> cr[0];
measure qr[1] -> cr[1];
measure qr[2] -> cr[2];
measure qr[3] -> cr[3];
measure qr[4] -> cr[4];
measure qr[5] -> cr[5];
measure qr[6] -> cr[6];
measure qr[7] -> cr[7];
measure qr[8] -> cr[8];
```

Current version of qibo is 0.2.7, qibolab is 0.1.6.
How can I solve this problem?

alecandido, 2024-05-18:
This is more an issue for Qibo than for Qibolab itself. Not the whole set of OpenQASM 3 is supported by Qibo, and there are no plans for doing it soon (especially beacuse of the classical instructions in QASM, that are specific for a certain way of doing things, not necessarily the one adopted by Qibo).

However, apart from the abstract explanation, here the problem is caused by the `barrier` instruction. You have two of them in your script:
```
barrier qr[0],qr[1],qr[2],qr[3],qr[4],qr[5],qr[6],qr[7],qr[8],qr[9];
```
These instructions are used to prevent optimization and simplification across the defined barrier.

No such a thing is happening in Qibo. So, simply drop the instructions from your code, and it could work (I did not implement it myself, nor I tested yet, but I do not see any other contrived instruction there).

alecandido, 2024-05-18:
Btw, @BrunoLiegiBastonLiegi, maybe this is a valid feedback for Qibo QASM module, i.e.:
- we could prune some instructions from scripts, that are essentially ineffective, and emit warning (if needed, in this case not even, because it will be automatically honored)
- for the remaining ones, we could try to emit better errors, detailing the offending instruction (instead of the AST class), if we're able to reconstruct it

However, I'm citing you to let you know, but they are definitely not a priority. So, I'd rather open an issue in Qibo listing the possible improvement.

closed 2024-07-16 by alecandido