qiboteam/qibolab #920: Qblox driver does not support couplers

aorgazf, 2024-06-20:
Hi,
Qblox driver in main does not work with routines that use coupler pulses.
I've introduced support for couplers to the old driver in alvaro/latest_20231215. 
I suppose it will not be difficult to port to main.

alecandido, 2025-04-16:
At this point, the answer is actually "yes".

I was waiting to merge the Qblox driver for 0.2, just because there was no alternative otherwise. But it's true that the process is advanced enough, that is practically sure that we do not want to spend this amount of effort on the 0.1 one. Which is also the only one for which this is issue is relevant.

stavros11, 2025-04-16:
Thanks for confirming. It also doesn't make sense to have a 0.1 related issue to a milestone for a 0.2 release. We could make a 0.1.* milestone and move such issues there, however, given that as you say, it is unlikely that we will put any effort in addressing them, it is just simpler to close and only keep those that could still be relevant for 0.2. The current one is certainly not.

closed 2025-04-16 by stavros11