qiboteam/qibolab #921: Qblox driver support for baking Custom pulses

aorgazf, 2024-06-20:
Hi,
Qblox driver does not support sweeping the duration of Custom pulses in real time. While we wait for Qblox to release the upcoming firmware that supports pre-distortion filters, Custom pulses are the only way to apply predistortions. These pulses are essential for the characterisation of 2q gates, especially those that use couplers.
This, and other enhancements to Custom pulses were introduced in the branch alvaro/latest_20231215.

DavidSarlle, 2024-07-31:
@hay-k @aorgazf @stavros11 @alecandido  we need this funcionality working on main in order to test the CZs and iSWAPS with qblox and iqm5q. 

I have created the runcard with the latest characterization obtained with older driver of qblox (because of https://github.com/qiboteam/qibolab/issues/919), but 2 qubit interaction routines from qibocal can not be tested without Custom pulses and this issue implemented.

hay-k, 2024-08-19:
I am not sure I correctly understand the scope of this request.

> Qblox driver does not support sweeping the duration of Custom pulses in real time.

> This, and other enhancements to Custom pulses were introduced in the branch alvaro/latest_20231215.

I couldn't find any commits in the mentioned branch that are related to doing real-time sweeps of custom pulse shapes in qblox. The only thing related to custom pulse shape is [this commit](https://github.com/qiboteam/qibolab/commit/e733b8f2e20c31c0178faedd5fb1215e3612cc83), which has nothing to do with qblox - it just fixes the deserialization of custom pulse shape from a string. The issue basically was that one couldn't read custom pulse shape from `parameters.yml` file, regardless of what instrument one uses, since this code is not instrument specific.

Is there anything else that is broken in relation to custom pulse shapes?

P.S. There were some changes in qibolab since the branch was forked, but I discovered that the deserialization of custom pulses still does not work. So the above linked PR fixes it in [this commit](https://github.com/qiboteam/qibolab/pull/993/commits/abadd83f41263a4211f44a1877889d948de0ada9).

DavidSarlle, 2024-08-19:
@hay-k I am not sure, but probably main, as you said, fixed the problems found by @aorgazf regarding the duration sweepers. But this si something that @aorgazf  should comment. 

What I know, is that in main, right now the custom pulses needed in qblox (no predistorsions to the pulses can be applied in the current firmaware) are needed for running 2q gates, predistorting flux pulses for example. For that, @aorgaz introduced some changes in the qblox driver to accept the custom shape:

"shape": 
"Custom([0.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.99934855, 0.9987093, 0.99808201, 0.99746647, 0.99686246, 0.99626975, 0.99568814, 0.99511741, 0.99455737, 0.99400782, 0.99346855, 0.99293938, 0.99242012, 0.99191058, 0.99141058, 0.99091994, 0.99043848, 0.98996604, 0.98950244, 0.98904752, 0.98860112, 0.98816308, 0.98773323, 0.98731144, 0.98689754, 0.98649139, 0.98609284, 0.98570176, 0.98531799, 0.98494141, 0.98457189, 0.98420927, 0.98385345, 0.98350429, 0.98316167, 0.98282546, 0.98249554, 0.98217181, 0.98185413, 0.9815424, 0.9812365, 0.98093634, 0.98064179, 0.98035276, 0.98006913, 0.97979082, 0.97951772, 0.97924973, 0.97898676, 0.97872871, 0.97847549, 0.97822702, 0.97798319, 0.97774393, 0.97750915, 0.97727876, 0.97705269, 0.97683085, 0.97661316, 0.97639955, 0.97618994, 0.97598425, 0.97578242, 0.97558436, 0.97539001, 0.9751993, 0.97501215, 0.97482852, 0.97464832, 0.97447149, 0.97429797, 0.97412771, 0.97396063, 0.97379667, 0.97363579, 0.97347792, 0.97332301, 0.97317099, 0.97302182, 0.97287545, 0.97273181, 0.97259087, 0.97245256, 0.97231684, 0.97218366, 0.97205298, 0.97192474, 0.9717989, 0.97167542, 0.97155425, 0.97143535, 0.97131868, 0.97120419, 0.97109184, 0.9709816, 0.97087342, 0.97076726, 0.97066309, 0.97056088, 0.97046057, 0.97036215, 0.97026557, 0.97017079, 0.97007779, 0.96998653, 0.96989698, 0.96980911, 0.96972288, 0.96963826, 0.96955523, 0.96947376, 0.96939381, 0.96931535, 0.96923837, 0.96916282, 0.96908869, 0.96901595, 0.96894457, 0.96887453, 0.9688058, 0.96873835, 0.96867217, 0.96860722, 0.9685435, 0.96848096, 0.9684196, 0.96835938, 0.96830029, 0.96824231, 0.96818541, 0.96812958, 0.9680748, 0.96802104, 0.96796828, 0.96791652, 0.96786572, 0.96781588, 0.96776696, 0.96771897, 0.96767187, 0.96762565, 0.9675803, 0.9675358, 0.96749213, 0.96744928, 0.96740723, 0.96736596, 0.96732547, 0.96728574, 0.96724676, 0.9672085, 0.96717096, 0.96713412, 0.96709797, 0.9670625, 0.96702769, 0.96699353, 0.96696001, 0.96692712, 0.96689485, 0.96686318, 0.9668321, 0.96680161, 0.96677168, 0.96674232, 0.96671351, 0.96668523, 0.96665749, 0.96663026, 0.0, 0.0, 0.0, 0.0, 0.0])",

but in main, this is not working, mainly because the function "eval" in the pulse class is not de-serializing properly the string expression of the pulse:

    @staticmethod
    def eval(value: str) -> "PulseShape":
        """Deserialize string representation.

        .. todo::

            To be replaced by proper serialization.
        """
        shape_name = re.findall(r"(\w+)", value)[0]
        if shape_name not in globals():
            raise ValueError(f"shape {value} not found")
        shape_parameters = re.findall(r"[-\w+\d\.\d]+", value)[1:]
        # TODO: create multiple tests to prove regex working correctly
        return globals()[shape_name](*shape_parameters)

hay-k, 2024-08-22:
yeah, this is exactly what I have fixed in the pull request. But this fix is purely related to the custom shape itself, and has nothing to do with the qblox driver, i.e. custom pulse deserialization does not work in main regardless of the instruments that you have. So I am wondering if the formulation "Qblox driver does not support ..." means there is also some additional problem that is qblox driver specific, or it was just non-optimal wording. @aorgazf ?