qiboteam/qibolab #934: Wrong qubits for two qubit gates in dummy

andrea-pasquale, 2024-07-15:
When I was testing https://github.com/qiboteam/qibocal/pull/938 I noticed that some of the two qubit gates for `dummy`
have the wrong qubits for the pulse shifting the phase.
https://github.com/qiboteam/qibolab/blob/d2c5f890faec916f364ac99f2e1e341efa7ab4a2/src/qibolab/dummy/parameters.json#L283
https://github.com/qiboteam/qibolab/blob/d2c5f890faec916f364ac99f2e1e341efa7ab4a2/src/qibolab/dummy/parameters.json#L399
https://github.com/qiboteam/qibolab/blob/d2c5f890faec916f364ac99f2e1e341efa7ab4a2/src/qibolab/dummy/parameters.json#L429
https://github.com/qiboteam/qibolab/blob/d2c5f890faec916f364ac99f2e1e341efa7ab4a2/src/qibolab/dummy/parameters.json#L479

It is not a big issue, however we should try to be consistent.
I can take care of opening the PR.

alecandido, 2024-08-08:
@andrea-pasquale if you wish to fix this, it would be appreciated.

You can base your PR directly on https://github.com/qiboteam/qibolab/tree/0.2 (unless it is merged before, of course)

andrea-pasquale, 2024-08-12:
It seems that all the bugs that I spotted are already fixed in 0.2.
Shall I close this @alecandido ?

closed 2024-08-16 by andrea-pasquale