qiboteam/qibolab #951: Complex numbers in samples and frequencies

marekgluza, 2024-08-06:
We have been submitting jobs to qw5q_platinum using the latest qibolab release 0.1.8 (`module load qibo`)

```python
import qibo
c = qibo.Circuit(5)
c.add(qibo.gates.GPI2(0,0))
c.add(qibo.gates.M(0,1,2,3,4))
result = c(nshots=100)
print(result.samples())
```

when we switch to frequencies it also fails because the results variables are complex

marekgluza, 2024-08-06:
`probabilities` also catches that

```
Traceback (most recent call last):
  File "/nfs/users/marek.gluza/h.py", line 26, in <module>
    print(result.probabilities())
  File "/nfs/tools/qibo/lib/python3.10/site-packages/qibo/result.py", line 288, in probabilities
    probs[state] = freq / self.nshots
TypeError: list indices must be integers or slices, not complex
```

stavros11, 2024-08-06:
Thanks @marekgluza for reporting this. I could reproduce.

After a quick investigation it seems to be related to the latest release of qibo (0.2.10). If you create your own environment and force the previous release (`pip install qibo==0.2.9`) the issue should not appear. I will check the [list of PRs](https://github.com/qiboteam/qibo/releases/tag/v0.2.10) in the latest release to see if I can identify which change caused this and update if I find something.

alecandido, 2024-08-06:
Thanks @marekgluza for reporting, and @stavros11 for the investigation. I thought for a moment that this could be related to the Rust failures, but unless it's a problem in the dependencies, it should be unrelated (since `execute_qasm()` should not pass through `qibo`, unless for just this part:
https://github.com/qiboteam/qibolab/blob/4f1459444a88666bad3b360dd728b0b88fa9c366/src/qibolab/backends.py#L74-L75
and the `MeasurementOutcome` object.

In any case, @stavros11, if you find anything new, please publish it here (in case of greater issues, I will also take a look at some point).