qiboteam/qibolab #952: Error handling for requesting two qubit gates

sorewachigauyo, 2024-08-06:
When the user requests a two-qubit gate, if it has not been calibrated or if the connectivity does not exist, the error message is given, e.g. `"Calibration for CZ gate between qubits 0 and 1 not found."`. If the user requests an invalid pairing that doesn't exist on the qubit device, a different message should be displayed to inform that the connectivity doesn't exist. This should be caught by the transpiler, but maybe this adjustment would be more user-friendly.

https://github.com/qiboteam/qibolab/blob/main/src/qibolab/platform/platform.py#L347-L354

alecandido, 2024-08-06:
Thanks @sorewachigauyo. This is an easy fix and a useful UX improvement.

However, I consider not to be that urgent. Thus, as we are focusing on 0.2 series, I'll move directly there.

alecandido, 2024-11-28:
Ok, now that we don't have any longer an independent topology, this may be slightly affected.

In any case, in principle we still an effective connectivity, given by the registered pairs. So, we should still be able to give two different error messages whether the pair is not registered at all, or if the gate is not calibrated at all (and thus `None`).

However, I would expect something different to be already happening anyhow. So, I'm not sure whether there is actually any action to be taken at all. But this statement has at least to be verified.

stavros11, 2024-12-01:
> However, I would expect something different to be already happening anyhow. So, I'm not sure whether there is actually any action to be taken at all. But this statement has at least to be verified.

To continue on this, what happens now can be easily verified with:
```py
from qibolab import create_platform

platform = create_platform("dummy")
platform.natives.two_qubit[(0, 1)]
```
In that case there is no gate registered for `(0, 1)` pair and the following is returned
```py
Traceback (most recent call last):
  File "/home/stavros/qiboteam/qibolab/src/qibolab/_core/parameters.py", line 84, in __getitem__
    return super().__getitem__(key)
           ^^^^^^^^^^^^^^^^^^^^^^^^
KeyError: (0, 1)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/stavros/qiboteam/qibolab/src/qibolab/_core/parameters.py", line 86, in __getitem__
    value = super().__getitem__((key[1], key[0]))
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
KeyError: (1, 0)
```
which is not particularly insightful so maybe could be improved a bit.

I would assume that if `(0, 1)` (or `(1, 0)`) existed as a key in `parameters.json` but had no gates registered to it, it would return an empty `TwoQubitNatives` container where all attributes are `None`.

As said above, there is no more `platform.connectivity`, so there is only one error case to handle, when the pair does not exist in two qubit natives (as above). There is also `platform.pairs` but this is just returning the keys of two qubit natives so there isn't much to do about it.

alecandido, 2024-12-02:
Yes, so the errors should be clearly distinguished. However, I would then change the issue into a broader one, since we then need a clear policy for where to recast the errors, and when to do that.

In a sense, this should be another consequence of the public API, since exceptions are also part of the API itself.