qiboteam/qibolab #975: Error in `Custom.__init__()`

csookim, 2024-08-13:
I tried running the following code using Slurm:

```py
from qibo import gates
from qibo.models.circuit import Circuit
from qibo.transpiler.pipeline import Passes
from qibo.transpiler.unroller import NativeGates, Unroller

circuit = Circuit(3)
circuit.add(gates.H(0))
circuit.add(gates.SWAP(0, 2))

ng = NativeGates.default()
custom_pipeline = Passes([Unroller(native_gates=ng)])
transpiled_circuit, _ = custom_pipeline(circuit)

print("Before unrolling:")
print(circuit.draw())
print("After unrolling:")
print(transpiled_circuit.draw())

results = transpiled_circuit(nshots=100)
results.state()
```

However, I got this error:

```
[Qibo 0.2.11|INFO|2024-08-13 08:40:31]: Loading platform /nfs/tools/qibolab_platforms_qrc/iqm5q
[Qibo 0.2.11|INFO|2024-08-13 08:40:31]: Using qibolab (/nfs/tools/qibolab_platforms_qrc/iqm5q) backend on /CPU:0
Before unrolling:
q0: ─H─x─
q1: ───|─
q2: ───x─
After unrolling:
q0: ─Z─GPI2─o─Z─GPI2─Z─Z─GPI2─o────────
q1: ────────|────────|────────|────────
q2: ─Z─GPI2─Z─Z─GPI2─o─Z─GPI2─Z─Z─GPI2─
Traceback (most recent call last):
  File "/nfs/users/changsoo.kim/test.py", line 20, in <module>
    results = transpiled_circuit(nshots=100)
  File "/nfs/tools/qibo/lib/python3.10/site-packages/qibo/models/circuit.py", line 1113, in __call__
    return self.execute(initial_state=initial_state, nshots=nshots)
  File "/nfs/tools/qibo/lib/python3.10/site-packages/qibo/models/circuit.py", line 1109, in execute
    return GlobalBackend().execute_circuit(self, initial_state, nshots)
  File "/nfs/tools/qibo/lib/python3.10/site-packages/qibolab/backends.py", line 100, in execute_circuit
    sequence, measurement_map = self.compiler.compile(circuit, self.platform)
  File "/nfs/tools/qibo/lib/python3.10/site-packages/qibolab/compilers/compiler.py", line 156, in compile
    gate_sequence, gate_phases = self._compile_gate(
  File "/nfs/tools/qibo/lib/python3.10/site-packages/qibolab/compilers/compiler.py", line 107, in _compile_gate
    gate_sequence, gate_phases = rule(gate, platform)
  File "/nfs/tools/qibo/lib/python3.10/site-packages/qibolab/compilers/default.py", line 88, in cz_rule
    return platform.create_CZ_pulse_sequence(gate.qubits)
  File "/nfs/tools/qibo/lib/python3.10/site-packages/qibolab/platform/platform.py", line 354, in create_CZ_pulse_sequence
    return self.pairs[pair].native_gates.CZ.sequence(start)
  File "/nfs/tools/qibo/lib/python3.10/site-packages/qibolab/native.py", line 248, in sequence
    sequence.add(pulse.pulse(start=start))
  File "/nfs/tools/qibo/lib/python3.10/site-packages/qibolab/native.py", line 82, in pulse
    return FluxPulse(
  File "/nfs/tools/qibo/lib/python3.10/site-packages/qibolab/pulses.py", line 1180, in __init__
    super().__init__(
  File "<string>", line 13, in __init__
  File "/nfs/tools/qibo/lib/python3.10/site-packages/qibolab/pulses.py", line 790, in __post_init__
    self.shape = PulseShape.eval(self.shape)
  File "/nfs/tools/qibo/lib/python3.10/site-packages/qibolab/pulses.py", line 223, in eval
    return globals()[shape_name](*shape_parameters)
TypeError: Custom.__init__() takes from 2 to 3 positional arguments but 181 were given
```

alecandido, 2024-08-13:
I believe this is duplicating #921

csookim, 2024-08-13:
I also got this error. Is there a problem with my code?

```py
[Qibo 0.2.11|INFO|2024-08-13 10:54:23]: Loading platform /nfs/tools/qibolab_platforms_qrc/qw5q_platinum
[Qibo 0.2.11|INFO|2024-08-13 10:54:23]: Using qibolab (/nfs/tools/qibolab_platforms_qrc/qw5q_platinum) backend on /CPU:0
[Qibo 0.2.11|INFO|2024-08-13 10:54:23]: Connecting to instrument <qibolab.instruments.zhinst.executor.Zurich object at 0x7c7b2026c460>.
Before unrolling:
q0: ─H─x─
q1: ───|─
q2: ───x─
After unrolling:
q0: ─Z─GPI2─o─Z─GPI2─Z─Z─GPI2─o────────
q1: ────────|────────|────────|────────
q2: ─Z─GPI2─Z─Z─GPI2─o─Z─GPI2─Z─Z─GPI2─
[2024.08.13 10:54:23.919] INFO    Logging initialized from [Default inline config in laboneq.laboneq_logging] logdir is /nfs/users/changsoo.kim/laboneq_output/log
[2024.08.13 10:54:23.920] INFO    VERSION: laboneq 2.25.0
[2024.08.13 10:54:23.920] INFO    Connecting to data server at localhost:8004
[2024.08.13 10:54:23.940] INFO    Connected to Zurich Instruments LabOne Data Server version 24.01 at localhost:8004
[2024.08.13 10:54:24.101] WARNING HDAWG:dev8660: Include the device options 'HDAWG8/' in the device setup ('options' field of the 'instruments' list in the device setup descriptor). This will become a strict requirement in the future.
[2024.08.13 10:54:24.629] WARNING SHFQC/QA:dev12146: Include the device options 'SHFQC/QC6CH' in the device setup ('options' field of the 'instruments' list in the device setup descriptor). This will become a strict requirement in the future.
[2024.08.13 10:54:25.963] INFO    Configuring the device setup
[2024.08.13 10:54:26.067] INFO    The device setup is configured
[Qibo 0.2.11|INFO|2024-08-13 10:54:26]: Connecting to instrument <qibolab.instruments.dummy.DummyLocalOscillator object at 0x7c7b2026c7f0>.
[Qibo 0.2.11|INFO|2024-08-13 10:54:26]: Connecting to instrument <qibolab.instruments.dummy.DummyLocalOscillator object at 0x7c7b2026e800>.
[Qibo 0.2.11|INFO|2024-08-13 10:54:26]: Connecting to instrument <qibolab.instruments.rohde_schwarz.SGS100A object at 0x7c7b2026e380>.
[Qibo 0.2.11|INFO|2024-08-13 10:54:26]: Connecting to instrument <qibolab.instruments.dummy.DummyLocalOscillator object at 0x7c7b2026e8c0>.
[Qibo 0.2.11|INFO|2024-08-13 10:54:26]: Connecting to instrument <qibolab.instruments.dummy.DummyLocalOscillator object at 0x7c7b2026e110>.
[Qibo 0.2.11|INFO|2024-08-13 10:54:26]: Minimal execution time (sequence): 0.0400237
Connected to: Rohde&Schwarz SGS100A (serial:1416.0505k02/113301, firmware:4.2.366.0-5.00.232.70) in 0.24s
Traceback (most recent call last):
  File "/nfs/users/changsoo.kim/test.py", line 19, in <module>
    results = transpiled_circuit(nshots=100)
  File "/nfs/tools/qibo/lib/python3.10/site-packages/qibo/models/circuit.py", line 1113, in __call__
    return self.execute(initial_state=initial_state, nshots=nshots)
  File "/nfs/tools/qibo/lib/python3.10/site-packages/qibo/models/circuit.py", line 1109, in execute
    return GlobalBackend().execute_circuit(self, initial_state, nshots)
  File "/nfs/tools/qibo/lib/python3.10/site-packages/qibolab/backends.py", line 105, in execute_circuit
    readout = self.platform.execute_pulse_sequence(
  File "/nfs/tools/qibo/lib/python3.10/site-packages/qibolab/platform/platform.py", line 196, in execute_pulse_sequence
    return self._execute(sequence, options, **kwargs)
  File "/nfs/tools/qibo/lib/python3.10/site-packages/qibolab/platform/platform.py", line 170, in _execute
    new_result = instrument.play(
  File "/nfs/tools/qibo/lib/python3.10/site-packages/qibolab/instruments/zhinst/executor.py", line 395, in play
    return self.sweep(qubits, couplers, sequence, options)
  File "/nfs/tools/qibo/lib/python3.10/site-packages/qibolab/instruments/zhinst/executor.py", line 716, in sweep
    self.experiment_flow(qubits, couplers, sequence, options)
  File "/nfs/tools/qibo/lib/python3.10/site-packages/qibolab/instruments/zhinst/executor.py", line 386, in experiment_flow
    self.sub_sequences, self.unsplit_channels = self.create_sub_sequences(
  File "/nfs/tools/qibo/lib/python3.10/site-packages/qibolab/instruments/zhinst/executor.py", line 356, in create_sub_sequences
    start, _ = measurement_start_end[measurement_index]
KeyError: 0

```

alecandido, 2024-08-13:
> I also got this error. Is there a problem with my code?

Not necessarily. The two problems are unrelated to each other.
Before you were not able to pass the compiler, since there was a problem with the natives.
Now the problem seems related to measurements: the Zurich driver seems to expect at least one measurement, and there are none in your circuit. Assuming I interpreted correctly, I guess the error could be improved (and raised earlier).

In any case, Qibolab is not that stable, and mainly supporting calibration. Sharp edges for circuit execution should be expected (even because it doesn't depend only on Qibolab, but even on the platforms' definition and calibration itself - which is much work-in-progress).

Try to add measurements to your circuit, and check whether you could receive any result. Do not expect to reproduce simulation, especially if 2-qubit gates are involved. But you should receive some shots.

Btw: you suspected an issue, and that was correct. However, if you're unsure about what is happening, and you'd like to ask for some support, discussions are better place.
Though they are currently disabled. Could we open them for Qibolab as well? @scarrazza

scarrazza, 2024-08-13:
Sure, I have just enabled it.

csookim, 2024-08-14:
Thanks, @alecandido. I got the result after adding the measurements.

alecandido, 2024-08-14:
@stavros11 in principle, we should have failed right after compilation
https://github.com/qiboteam/qibolab/blob/4f1459444a88666bad3b360dd728b0b88fa9c366/src/qibolab/backends.py#L100-L108
as soon as we know the `measurement_map`, and we know it's empty. Not inside compilation, because it would not be modular (it may still be useful to compile a circuit without measurements, e.g. for testing).

If you agree, I would open a dedicated issue, and close this one, since it's not very specific (and the Qblox issue is already documented, as discussed before).