qiboteam/qibolab #987: Gate to pulse scheduling

sorewachigauyo, 2024-08-16:
I have a sample code as follows

```python
import qibo
qibo.set_backend("qibolab", platform="dummy")

from qibo.gates import RZ, GPI2, iSWAP, M
import numpy as np

circ = qibo.Circuit(2)
# Hadamard
circ.add(RZ(0, np.pi))
circ.add(GPI2(0, 0))
circ.add(RZ(0, np.pi / 2))
circ.add(GPI2(0, -np.pi))

# CNOT (partial)
circ.add(GPI2(1, 0))
circ.add(M(0, 1))

res = circ(nshots=1000)
```

I would expect three pulses, with the two pulses on qubit 0 and qubit 1 done in parallel and the third pulse on qubit 0 to be done just after.

However, the pulse sequence I get is
```
DrivePulse(0, 40, 0.15, 4_200_000_000.0, 0, Drag(5, -0.02), drive-1, 1)
DrivePulse(40, 40, 0.05, 4_000_000_000.0, 3.141593, Gaussian(5), drive-0, 0)
DrivePulse(80, 40, 0.05, 4_000_000_000.0, 1.570796, Gaussian(5), drive-0, 0)
ReadoutPulse(120, 2000, 0.1, 5_200_000_000.0, 0, GaussianSquare(5, 0.75), readout, 0)
ReadoutPulse(120, 2000, 0.1, 4_900_000_000.0, 0, GaussianSquare(5, 0.75), readout, 1)
```

Here, the pulses are done sequentially, which can lead to unnecessary bloating of the circuit depth.

On Qibolab main branch commit `9bbbbe8c` v0.1.9
On QIbo main branch commit `4f57f37` v0.2.12

alecandido, 2024-08-16:
> Here, the pulses are done sequentially, which can lead to unnecessary bloating of the circuit depth.

This has been solved in #965, thus it is already fixed in https://github.com/qiboteam/qibolab/tree/0.2

There is already one remaining issue for explicit two-qubit gates https://github.com/qiboteam/qibolab/issues/976 (also present in 0.1), but there are none in your script.

Could you try with the new version?

sorewachigauyo, 2024-08-16:
Sure, the delays for the readout channels are appropriate.

```python
[[('qubit_0/drive', VirtualZ(phase=3.141592653589793)), ('qubit_1/drive', Pulse(duration=40.0, amplitude=0.15, envelope=Drag(kind='drag', rel_sigma=5.0, beta=0.02), relative_phase=0.0)), ('qubit_0/drive', Pulse(duration=40.0, amplitude=0.05, envelope=Gaussian(kind='gaussian', rel_sigma=5.0), relative_phase=0.0)), ('qubit_0/drive', VirtualZ(phase=1.5707963267948966)), ('qubit_0/drive', Pulse(duration=40.0, amplitude=0.05, envelope=Gaussian(kind='gaussian', rel_sigma=5.0), relative_phase=3.141592653589793)), ('qubit_0/probe', Delay(duration=80.0)), ('qubit_1/probe', Delay(duration=80.0)), ('qubit_0/probe', Pulse(duration=2000.0, amplitude=0.1, envelope=GaussianSquare(kind='gaussian_square', rel_sigma=5.0, width=0.75), relative_phase=0.0)), ('qubit_1/probe', Pulse(duration=2000.0, amplitude=0.1, envelope=GaussianSquare(kind='gaussian_square', rel_sigma=5.0, width=0.75), relative_phase=0.0))]]
```