qiboteam/qibolab #1002: Result dump does not store samples

scarrazza, 2024-08-23:
I am trying to dump the circuit results object and reload. 
The qibolab backend is not storing samples therefore the code crashes.

Code example using the latest `module load qibo` and `sbatch -p qw5q_platinum`):
```python
import qibo
c = qibo.Circuit(5)
c.add(qibo.gates.M(0))
r = c(nshots=100)
r.dump("./output.npy")
```
Then, while loading:
```python
import qibo
qibo.result.load_result("output.npy").samples()
```
we get:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/nfs/users/test.test/qibo/src/qibo/result.py", line 341, in samples
    samples = self.backend.sample_shots(self._probs, self.nshots)
  File "/nfs/users/test.test/qibo/src/qibo/backends/numpy.py", line 650, in sample_shots
    range(len(probabilities)), size=nshots, p=probabilities
TypeError: object of type 'NoneType' has no len()
```

alecandido, 2024-08-23:
The problem (that applies both to 0.1 and 0.2) is not related to the dump operation, since `QibolabBackend` is not doing anything specific
https://github.com/qiboteam/qibolab/blob/0.1/src/qibolab/backends.py
nor the general Qibolab is forgetting the measurements: they are assigned to the `MeasurementObject`
https://github.com/qiboteam/qibolab/blob/bbac7cf23e0d879f792a16f64e2f0f5e7b147a01/src/qibolab/backends.py#L112-L114
and then it is its responsibility to dump and load its own content.

Before dumping, can you access the measurements from the object `r`? (i.e. `r.measurements`)
In any case, I will try to debug the problem.

alecandido, 2024-08-23:
Ok, no, I see the problem: the shots are in multiple places.

The `MeasurementResult` class owns the shots:
https://github.com/qiboteam/qibo/blob/386fe25c69e54211b4816702b5f0b619d645b35c/src/qibo/measurements.py#L88
and it is owned on its own by the `M` gate
https://github.com/qiboteam/qibo/blob/386fe25c69e54211b4816702b5f0b619d645b35c/src/qibo/gates/measurements.py#L59

Qibolab is adding the shots there:
https://github.com/qiboteam/qibolab/blob/bbac7cf23e0d879f792a16f64e2f0f5e7b147a01/src/qibolab/backends.py#L59-L75

Instead, there is another class, the `MeasurementOutcome`, that collects all the measurements in a circuit, has its own shots in the `._samples` attribute:
https://github.com/qiboteam/qibo/blob/386fe25c69e54211b4816702b5f0b619d645b35c/src/qibo/result.py#L181
it also has a complex `.samples()` method, computing the shots, possibly from the gates
https://github.com/qiboteam/qibo/blob/386fe25c69e54211b4816702b5f0b619d645b35c/src/qibo/result.py#L303-L372

However, the `.dump()` operation is then using the bare attribute `._samples`
https://github.com/qiboteam/qibo/blob/386fe25c69e54211b4816702b5f0b619d645b35c/src/qibo/result.py#L419

To me, the solution should not be propagating the complication of the double measurement registration to the `QibolabBackend`, but just call `self.samples()` in `MeasurementOutcome.to_dict()` in Qibo.

@scarrazza, if you agree, I would transfer the issue.

scarrazza, 2024-08-23:
Thanks @alecandido for debugging, please go ahead with the transfer issue. I agree with the solution you propose.