qiboteam/qibolab #1003: Minor native & sequence improvements

alecandido, 2024-08-24:
The idea is to make it simpler to create sequences.

Currently, you have to do something like this:
```py
qd_seq = natives.RX.create_sequence()
probe_seq = natives.MZ.create_sequence()
sequence = PulseSequence()
sequence.concatenate(qd_seq) 
sequence.concatenate(probe_seq)
```
(cf. `conftest.py`, the `execute` fixture)

A couple of proposals to make it simpler are:
- [ ] alias `Native.create_sequence()` with `.__call__()`
- [ ] alias `PulseSequence.concatenate()` with `.__or__()`
    - the true alias for `.concatenate()` will be `.__ior__()`, since the operation is in-place
    - but we can make `s1 | s2` just by creating a new sequence:
        ```py
        s = PulseSequence()
        s |= s1
        s |= s2
        ```

It should be possible to then write it like the following:
```py
q0 = natives
sequence = q0.RX() | q0.MZ()
```

stavros11, 2024-08-30:
@alecandido following the meeting last Wednesday, I understood that `PulseSequence.concatenate` in the last example pads the the sequence with delays so that the `MZ` plays after `RX`. However, the following example:
```py
from qibolab import create_platform
from qibolab.sequence import PulseSequence

def sequence_print(sequence: PulseSequence):
    for ch, pulse in sequence:
        print(ch, pulse)
    print()

platform = create_platform("dummy")

q0 = platform.natives.single_qubit[0]
rx_sequence = q0.RX.create_sequence()
mz_sequence = q0.MZ.create_sequence()

sequence = PulseSequence()
sequence.concatenate(rx_sequence)
sequence.concatenate(mz_sequence)

sequence_print(rx_sequence)
sequence_print(mz_sequence)
sequence_print(sequence)

```
shows that there are no delays present in `sequence`.

I think we need to replace https://github.com/qiboteam/qibolab/blob/888753403e496f843d67dd398b3f8ec424ee3749/src/qibolab/sequence.py#L106

with 
```py
_synchronize(self, self.channels | other.channels) 
```
in order to get the delays.

alecandido, 2024-08-30:
Yes, that's correct. I can add this change in the PR that will implement this issue.

Two *concatenated* `PulseSequence` should actually start one after the other. If they need to be played in parallel (as it could be the case for different qubits) you should instead use `.extend()` (or the associated list operator, i.e. `+`, which will create a new sequence, same as `|` will do for `.concatenate()`).

closed 2024-09-13 by alecandido