qiboteam/qibolab #1024: Update QM driver for OPX1000

stavros11, 2024-09-03:
From what I understood this needs to be done for both 0.1 and 0.2.

alecandido, 2024-09-03:
> From what I understood this needs to be done for both 0.1 and 0.2.

Well, we can do for 0.1, but it greatly depends on the timing, especially the transition of Qibocal to 0.2.
For the time being, we never made use of the OPX1000, as it was not connected to any chip. As soon as we'll need to operate it, it will have to be updated in the driver version used. So:
- if it will be after Qibocal update to 0.2, we only need a 0.2 update
- if it will be much before, we only need a 0.1 update (for the time being)
- if it will happen to be available more or less during the transition, we'll need both more or less at the same time

alecandido, 2024-09-03:
In any case, I confirm that at the API level nothing should change on our side (i.e. in the `program` part), other than whatever will be required by the version update.

Regarding the configurations, it seems that there are actually three cases (not just OPX+ and OPX1000) and one of the additional ones (the LF) only adds a "FEM-scope" on top of the usual configs, while the MW also requires a different ports' definition.
https://docs.quantum-machines.co/1.2.0/docs/Introduction/config/
https://docs.quantum-machines.co/1.2.0/docs/Guides/opx1000_fems/

stavros11, 2024-09-03:
Well, for sure we need for 0.2 at some point, most likely either 0.2.0 or 0.2.1. For 0.1, it depends on how soon we need it and if 0.2 is ready before that. From what I understood from the meeting, they need it immediately to use with qibocal and 0.2 is for sure not ready for that. Therfore, I am pretty sure we need to update both versions.

As for the changes, I also think we only need to change the config.

alecandido, 2024-09-03:
> From what I understood from the meeting, they need it immediately to use with qibocal and 0.2 is for sure not ready for that. Therfore, I am pretty sure we need to update both versions.

They will need soon after the 1 week trial, starting whenever they are ready. Let's see.

> As for the changes, I also think we only need to change the config.

👌