qiboteam/qibolab #1029: QM driver in 0.1 fails to run `rabi_length` experiment

andrea-pasquale, 2024-09-05:
The following qibocal runcard 
```yaml
platform: qw11q
targets: [D1]

actions:

  - id: rabi
    operation: rabi_length_signal
    parameters:
        pulse_duration_start: 20
        pulse_duration_end: 40
        pulse_duration_step: 1
        pulse_amplitude: 0.06
        relaxation_time: 100_000
        nshots: 1024
```
fails with an error related to baking.
<details>
<summary>Error</summary>
<br>
[Qibo 0.2.10|INFO|2024-09-05 10:57:23]: Loading platform dummy
2024-09-05 10:57:24,433 - qm - INFO     - Starting session: 432b70c8-0470-4214-929d-7b51f5ac2ca4
2024-09-05 10:57:27,319 - qm - INFO     - Octave "octave2" Health check passed, current temperature 60
2024-09-05 10:57:28,053 - qm - INFO     - Octave "octave3" Health check passed, current temperature 57
2024-09-05 10:57:28,784 - qm - INFO     - Octave "octave4" Health check passed, current temperature 58
2024-09-05 10:57:29,604 - qm - INFO     - Octave "octave5" Health check passed, current temperature 59
2024-09-05 10:57:30,327 - qm - INFO     - Octave "octave6" Health check passed, current temperature 58
2024-09-05 10:57:30,327 - qm - INFO     - Performing health check
2024-09-05 10:57:30,347 - qm - WARNING  - Health check warning: Inter-OPX connectivity issues in OPX: con1. Missing ports are: 12, 11, 10, 9. See QM-App for more info.
2024-09-05 10:57:30,347 - qm - WARNING  - Health check warning: Inter-OPX connectivity issues in OPX: con2. Missing ports are: 12, 11, 10, 9. See QM-App for more info.
2024-09-05 10:57:30,347 - qm - WARNING  - Health check warning: Inter-OPX connectivity issues in OPX: con3. Missing ports are: 12, 11, 10, 9. See QM-App for more info.
2024-09-05 10:57:30,347 - qm - WARNING  - Health check warning: Inter-OPX connectivity issues in OPX: con4. Missing ports are: 12, 11, 10, 9. See QM-App for more info.
2024-09-05 10:57:30,347 - qm - WARNING  - Health check warning: Inter-OPX connectivity issues in OPX: con5. Missing ports are: 12, 11, 10, 9. See QM-App for more info.
2024-09-05 10:57:30,347 - qm - WARNING  - Health check warning: Inter-OPX connectivity issues in OPX: con6. Missing ports are: 12, 11, 10, 9. See QM-App for more info.
2024-09-05 10:57:30,347 - qm - WARNING  - Health check warning: Inter-OPX connectivity issues in OPX: con7. Missing ports are: 12, 11, 10, 9. See QM-App for more info.
2024-09-05 10:57:30,347 - qm - WARNING  - Health check warning: Inter-OPX connectivity issues in OPX: con8. Missing ports are: 12, 11, 10, 9. See QM-App for more info.
2024-09-05 10:57:30,347 - qm - WARNING  - Health check warning: Inter-OPX connectivity issues in OPX: con9. Missing ports are: 12, 11, 10, 9. See QM-App for more info.
2024-09-05 10:57:30,347 - qm - INFO     - Health check passed
Connected to: Rohde&Schwarz SGS100A (serial:1416.0505k02/114167, firmware:4.2.76.0-4.30.046.295) in 0.16s
Connected to: Rohde&Schwarz SGS100A (serial:1416.0505k02/114164, firmware:4.2.76.0-4.30.046.295) in 0.14s
Traceback (most recent call last):
  File "/nfs/users/andrea.pasquale/default/bin/qq", line 8, in <module>
    sys.exit(command())
  File "/nfs/users/andrea.pasquale/default/lib/python3.10/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/nfs/users/andrea.pasquale/default/lib/python3.10/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/nfs/users/andrea.pasquale/default/lib/python3.10/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/nfs/users/andrea.pasquale/default/lib/python3.10/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/nfs/users/andrea.pasquale/default/lib/python3.10/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/nfs/users/andrea.pasquale/qibocal/src/qibocal/cli/_base.py", line 75, in auto
    autocalibrate(runcard, folder, force, update)
  File "/nfs/users/andrea.pasquale/qibocal/src/qibocal/cli/autocalibration.py", line 42, in autocalibrate
    history = runcard.run(
  File "/nfs/users/andrea.pasquale/qibocal/src/qibocal/auto/runcard.py", line 63, in run
    instance.run_protocol(
  File "/nfs/users/andrea.pasquale/qibocal/src/qibocal/auto/execute.py", line 131, in run_protocol
    completed = task.run(platform=self.platform, targets=self.targets, mode=mode)
  File "/nfs/users/andrea.pasquale/qibocal/src/qibocal/auto/task.py", line 149, in run
    completed.data, completed.data_time = operation.acquisition(
  File "/nfs/users/andrea.pasquale/qibocal/src/qibocal/auto/operation.py", line 42, in wrapper
    out = func(*args, **kwds)
  File "/nfs/users/andrea.pasquale/qibocal/src/qibocal/protocols/rabi/length.py", line 88, in _acquisition
    results = platform.sweep(
  File "/nfs/users/andrea.pasquale/qibolab/src/qibolab/platform/platform.py", line 301, in sweep
    new_result = instrument.sweep(
[... 48 lines cut ...]
  File "/nfs/users/andrea.pasquale/qibocal/src/qibocal/auto/runcard.py", line 63, in run
    instance.run_protocol(
  File "/nfs/users/andrea.pasquale/qibocal/src/qibocal/auto/execute.py", line 131, in run_protocol
    completed = task.run(platform=self.platform, targets=self.targets, mode=mode)
  File "/nfs/users/andrea.pasquale/qibocal/src/qibocal/auto/task.py", line 149, in run
    completed.data, completed.data_time = operation.acquisition(
  File "/nfs/users/andrea.pasquale/qibocal/src/qibocal/auto/operation.py", line 42, in wrapper
    out = func(*args, **kwds)
  File "/nfs/users/andrea.pasquale/qibocal/src/qibocal/protocols/rabi/length_signal.py", line 92, in _acquisition
    results = platform.sweep(
  File "/nfs/users/andrea.pasquale/qibolab/src/qibolab/platform/platform.py", line 301, in sweep
    new_result = instrument.sweep(
  File "/nfs/users/andrea.pasquale/qibolab/src/qibolab/instruments/qm/controller.py", line 372, in sweep
    qmsequence, ro_pulses = self.create_sequence(qubits, sequence, sweepers)
  File "/nfs/users/andrea.pasquale/qibolab/src/qibolab/instruments/qm/controller.py", line 343, in create_sequence
    qmpulse.bake(self.config, durations=[pulse.duration])
  File "/nfs/users/andrea.pasquale/qibolab/src/qibolab/instruments/qm/sequence.py", line 165, in bake
    segment.add_op(self.operation, self.element, waveform)
  File "/nfs/users/andrea.pasquale/default/lib/python3.10/site-packages/qualang_tools/bakery/bakery.py", line 778, in add_op
    raise ValueError("Element provided does not have any analog input")
ValueError: Element provided does not have any analog input
srun: error: fahid: task 0: Exited with exit code 1
<br><br>
</details>
Is there something wrong with the parameters provided in the runcard?

alecandido, 2024-09-05:
Additional info: the relevant part of the error is just

```py
  File "/nfs/users/andrea.pasquale/qibolab/src/qibolab/instruments/qm/sequence.py", line 165, in bake
    segment.add_op(self.operation, self.element, waveform)
  File "/nfs/users/andrea.pasquale/default/lib/python3.10/site-packages/qualang_tools/bakery/bakery.py", line 778, in add_op
    raise ValueError("Element provided does not have any analog input")
ValueError: Element provided does not have any analog input
```

and of course it is related to 0.1.

It seems it's getting a misconfigured element, if not a completely wrong one...

alecandido, 2025-10-20:
The problem only regarded 0.1, which by now has been discontinued.

closed 2025-10-20 by alecandido