qiboteam/qibolab #1034: Should `Instrument` have a `name`?

stavros11, 2024-09-09:
In 0.2 we dropped the `name` attribute from `Qubit` and `Channel`, since it is stored in the corresponding maps in the `Platform`. However, `Instrument`s still have a `name` attribute, even though they are also stored in a map in the `Platform`. Unless we need `instrument.name` for something else that I am missing, we should remove `name` from instruments to be symmetric with the other objects.

@alecandido let me know if you agree and I will open a PR to remove it. If instead there is a good reason to keep, we can just close this issue.

This affects the interface, because instruments need to be used when writing platforms. Therefore we need to decide before 0.2.0.

alecandido, 2024-09-09:
> @alecandido let me know if you agree and I will open a PR to remove it. If instead there is a good reason to keep, we can just close this issue.

Agreed. If needed for internal purposes (within the driver), a name should be requested as a special attribute. If not needed, it should not be enforced, since objects should not know how they are referenced somewhere else (in a sense, the name of any object from its own point of view is `self`).

stavros11, 2024-09-13:
Fixed by #1037.

PR #1037 (2024-09-13): Drop instrument names
Closes #1034, if we finally decide to drop `Instrument.name`.

I also dropped the `OCTAVE_ADDRESS_OFFSET` from the QM driver, so that the user has to pass the full address when defining the instruments. Before we were adding a 11000 which I think is a default of Octaves, however if at some point this changes it would break our code, while now it is more flexible to potential changes.
closed 2024-09-13 by stavros11