qiboteam/qibolab #1048: Compiler not able to generate Bell state with non-native gates

andrea-pasquale, 2024-09-19:
Following #1044 I was able to generate a Bell circuit using native gates.

```py
from qibo import models, Circuit

circuit = Circuit(2)
circuit.add(gates.GPI2(0, phi=np.pi/2))
circuit.add(gates.GPI2(1, phi=np.pi/2))
circuit.add(gates.CZ(0, 1))
circuit.add(gates.X(0))
circuit.add(gates.GPI2(1, phi=np.pi/2))
circuit.add(gates.M(0,1))

print(circuit().frequencies())
```

However, when I try to run Bell using generic gates 

```py
from qibo import models, Circuit

circuit = Circuit(2)
circuit.add(gates.H(0))
circuit.add(gates.H(1))
circuit.add(gates.CZ(0, 1))
circuit.add(gates.H(1))
circuit.add(gates.M(0,1))

print(circuit().frequencies())
```

The results are completely wrong. We might need to debug it to understand if this issue is related to miscalibration or if there is still something to be fixed in the compiler.

jykhoo1987, 2024-09-25:
I think I encountered something similar as well when I was testing the emulator but back then I was still developing the features and I wasn't sure what was wrong. It might be worthwhile to repeat this on the default_q01_flux platform (https://github.com/qiboteam/qibolab/tree/1q_emulator_upgrades/tests/emulators/default_q01_flux) in 1q_emulator_upgrades branch to see if the problem is reproduced there as well. It might be helpful with troubleshooting, or at least confirming that the issue is not with miscalibration. I believe the emulator should be working correctly, but if you try this and encounter something wrong with the emulator do let me know so that I can fix it.

alecandido, 2024-09-25:
Neither the emulator nor the drivers are in doubt (not by me, at this stage), but rather the most relevant moving parts are the compiler, the transpiler, and (much less) the routines' definitions (boiling down to confirm that what we believe to be wrong, it's actually wrong).

In any case, the emulator is a useful tool to avoid hardware "noise" (metaphorically, i.e. including any kind of physical issue).
Right now, we're not short of hardware, and decently calibrated. But in general, a working emulator would be always available, in contrast to actual hardware...