qiboteam/qibolab #1058: Adjusting pulse duration with long decimal numbers using replace leads to sequence execution failure

sorewachigauyo, 2024-10-02:
Creating a simple sequence and changing the duration of the pulse with `replace` to a random number with some trailing digits fails some pydantic check. This may be a problem for some systems with irrational sampling intervals. Here is an example code on the dummy platform.

```python
import qibolab
from qibolab._core.serialize import replace

platform = qibolab.create_platform("dummy")
qubit = platform.qubits[0]
native = platform.natives.single_qubit[0]

chan, pulse = native.RX.create_sequence()[0]
new_pulse = replace(pulse, duration=22.3453)

sequence = qibolab.PulseSequence([(chan, new_pulse)])
sequence.concatenate(native.MZ.create_sequence())
platform.execute([sequence])
```

Here is the traceback
```
[Qibo 0.2.12|INFO|2024-10-03 01:29:03]: Loading platform dummy
[Qibo 0.2.12|INFO|2024-10-03 01:29:03]: Minimal execution time: 0.0020480000000000003
Traceback (most recent call last):
  File "c:\Users\pault\Documents\repos\qcs-test\test2.py", line 14, in <module>
    platform.execute([sequence])
  File "C:\Users\pault\Documents\repos\qibolab\src\qibolab\_core\platform\platform.py", line 285, in execute
    for b in batch(sequences, bounds):
  File "C:\Users\pault\Documents\repos\qibolab\src\qibolab\_core\unrolling.py", line 96, in batch
    update = Bounds.update(sequence)
             ^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\pault\Documents\repos\qibolab\src\qibolab\_core\unrolling.py", line 61, in update
    return cls(**up)
           ^^^^^^^^^
  File "C:\Users\pault\AppData\Local\Programs\Python\Python311\Lib\site-packages\pydantic\main.py", line 193, in __init__
    self.__pydantic_validator__.validate_python(data, self_instance=self)
pydantic_core._pydantic_core.ValidationError: 1 validation error for Bounds
waveforms
  Input should be a valid integer, got a number with a fractional part [type=int_from_float, input_value=23.3453, input_type=float]   
    For further information visit https://errors.pydantic.dev/2.8/v/int_from_float
```

Constructing a pulse from the same duration doesn't yield an error
```
import qibolab

platform = qibolab.create_platform("dummy")
qubit = platform.qubits[0]
native = platform.natives.single_qubit[0]

chan, pulse = native.RX.create_sequence()[0]
new_pulse = qibolab.Pulse(duration=22.3453, amplitude=1, envelope=qibolab.Rectangular())

sequence = qibolab.PulseSequence([(chan, new_pulse)])
sequence.concatenate(native.MZ.create_sequence())
platform.execute([sequence])
```

alecandido, 2024-10-02:
I will check it for sure, but this is truly weird, as the `duration` itself is supposed to be a `float`. That's why you're getting no error in the second case.

> ```
> pydantic_core._pydantic_core.ValidationError: 1 validation error for Bounds
> ```

The error is related to [`Bounds`](https://github.com/qiboteam/qibolab/blob/76a910760481620502b43af3a677078c5743b281/src/qibolab/_core/unrolling.py#L43), which are part of the unrolling infrastructure.

I will investigate the issue.

stavros11, 2024-10-03:
Thanks for reporting this @sorewachigauyo. It should be fixed by #1060, with some details about why it happens.

> Constructing a pulse from the same duration doesn't yield an error
> 
> ```python
> import qibolab
> 
> platform = qibolab.create_platform("dummy")
> qubit = platform.qubits[0]
> native = platform.natives.single_qubit[0]
> 
> chan, pulse = native.RX.create_sequence()[0]
> new_pulse = qibolab.Pulse(duration=22.3453, amplitude=1, envelope=qibolab.Rectangular())
> 
> sequence = qibolab.PulseSequence([(chan, new_pulse)])
> sequence.concatenate(native.MZ.create_sequence())
> platform.execute([sequence])
> ```

I think that if you change that to
```py
new_pulse = qibolab.Pulse(duration=22.3453, amplitude=1, envelope=qibolab.Gaussian(rel_sigma=0.2))
```
you will still get the error, for reasons explained in #1060. So it is not due to `replace`.

PR #1060 (2024-10-03): Float waveforms
Fixes #1058 (I was very curious so had to check). @alecandido, maybe you can also confirm, but I believe `Bounds.waveforms` should be `float` instead of `int` as it is calculated as a `sum(pulse.duration)` and `pulse.duration` is `float`.

The reason we did not catch this in the tests is because even though `duration` is annotated as `float`, we are only using integer durations in the dummy platform used in tests. Here I changed one of the durations to a float and indeed some tests were failing before the fix.

Also, the reason the second cell from https://github.com/qiboteam/q [... cut]
closed 2024-10-03 by stavros11