qiboteam/qibolab #1069: Sequences with different frequencies are not separated

jevillegasdatTII, 2024-10-09:
https://github.com/qiboteam/qibolab/blob/898fa86c2ed60f3a0cd149715c73db508f924f9b/src/qibolab/pulses.py#L1586

When sending two pulse sequences over the same channel, provided they don't overlap in time, they are not getting separated into different sequencers. Consequently, the frequency of the first pulse defines the **nco** frequency for the sequence and all the following pulses use the same frequency irrespective of their settings. For instance, the following code generates the attached sequences sent to qblox.

``` python
sequence = PulseSequence()
rx_pulse_q0 = platform.create_RX_pulse(0, start = 0)
rx_pulse_q1 = platform.create_RX_pulse(1, start = 0)
sequence.add(rx_pulse_q0)
sequence.add(rx_pulse_q1)
cr_pulse = platform.create_RX_pulse(1, start = rx_pulse_q1.finish)
cr_pulse.frequency = rx_pulse_q0.frequency
cr_pulse.duration = 1000
sequence.add(cr_pulse)
```

[Z_qcm_rf0_sequencer1_sequence.json](https://github.com/user-attachments/files/17304152/Z_qcm_rf0_sequencer1_sequence.json)
[Z_qcm_rf0_sequencer0_sequence.json](https://github.com/user-attachments/files/17304154/Z_qcm_rf0_sequencer0_sequence.json)
[Z_qcm_rf0_snapshot.json](https://github.com/user-attachments/files/17304155/Z_qcm_rf0_snapshot.json)

Notably, only two sequencers are created, where we need three (channel 2 with two different frequencies and channel one a single one).

jevillegasdatTII, 2024-10-09:
Possble fix in commit [f09f5a4](https://github.com/qiboteam/qibolab/commit/f09f5a4e1b6ac62e0dd4443eec3f7d7dd88e20a0)

hay-k, 2024-10-09:
@jevillegasdatTII I will take a look at your commits, but for the future can you please not push any commits directly to the branch qblox-ad-hoc-features, and instead create a PR pointing to that branch.

jevillegasdatTII, 2024-10-09:
Absolutely, sorry. :(