qiboteam/qibolab #1073: Use of `wire_names`

csookim, 2024-10-16:
`qibolab` already allows selecting target qubits using qubit names `str`. `Circuit.wire_names: list[str]` contains these qubit names in `qibo`. The method for accessing them needs to be updated.

https://github.com/qiboteam/qibo/issues/1493

csookim, 2024-10-23:
https://github.com/qiboteam/qibolab/blob/946a04f21f522cd05f898de91270153af7cdceff/src/qibolab/_core/compilers/compiler.py#L186

https://github.com/qiboteam/qibolab/blob/946a04f21f522cd05f898de91270153af7cdceff/src/qibolab/_core/compilers/compiler.py#L121

https://github.com/qiboteam/qibolab/blob/946a04f21f522cd05f898de91270153af7cdceff/src/qibolab/_core/compilers/compiler.py#L87

https://github.com/qiboteam/qibolab/blob/946a04f21f522cd05f898de91270153af7cdceff/src/qibolab/_core/platform/platform.py#L317-L339

alecandido, 2024-10-23:
Yes, as you noticed, the `str` names are already supported at the `Platform` level.

I confirm that you should just make use of them in the compiler, where you are receiving the `Circuit` directly from the backend.
https://github.com/qiboteam/qibolab/blob/946a04f21f522cd05f898de91270153af7cdceff/src/qibolab/_core/backends.py#L101

csookim, 2024-10-29:
Once https://github.com/qiboteam/qibo/issues/1493#issue-2591559398 is merged, all code using `_wire_names` should be updated.

RoyStegeman, 2026-11-02:
Resolved in #1083: the compiler now resolves qubits through `Circuit.wire_names`. The one remaining spot, where end-of-gate padding still used the logical index, is fixed in a #1626.

closed 2026-11-02 by RoyStegeman