qiboteam/qibolab #1087: Delay on gates

igres26, 2024-11-01:
All the gates have a "relative start" which I feel is barely used. 

However, when we calibrate Two-Qubit gates, we are allowed to play around with a delay **after** the gate is applied. It helps with the virtual phases for example. But there is no way to add a delay after a gate is applied into the parameters of the gate, making these routines that delay on this delay `dt` a bit useless.

Could we add on the create CZ module an input for a delay after the pulse is applied?

stavros11, 2024-11-01:
Thanks for reporting this. 

I don’t think we will address it as we avoid implementing new features for qibolab 0.1.

In 0.2 there is no `start` (and `relative_start`) in pulses anymore, as it has been replaced by the `Delay` object. Also, all native gates are defined as `PulseSequence`s which can also contain `Delay`s. Therefore, I would expect that the issue is already addressed there, even though it is not explicitly tested.

alecandido, 2024-11-01:
> I don’t think we will address it as we avoid implementing new features for qibolab 0.1.

Exactly.

> Therefore, I would expect that the issue is already addressed there, even though it is not explicitly tested.

We're generally lacking tests on hardware, especially for specific cases. But yes, once the `Delay` is there, it is just a matter of honoring it in the drivers' implementation, which could be already tested on QM (OPX+ and OPX1000). From the point of view of the general Qibolab support, the use case is just supported, since any `Delay` in a sequence will be taken into account as a valid delay.

Could we rely on you for testing it with QM? @igres26