qiboteam/qibolab #1093: KeyError: 'GPI'

scarrazza, 2024-11-03:
After installing qibolab 0.1.10, the the following code:
```python
import qibo
c = qibo.Circuit(17)
c.add(qibo.gates.GPI2(12,0))
c.add(qibo.gates.M(12))
result = c(nshots=100)
print(result.samples())
print(result.frequencies())
```
executed with `module load qibo` crashes on qw11q with:
```
2024-11-03 18:25:04,167 - qm - INFO     - Starting session: 903803b8-8d72-48e9-9dc3-0c438f6b5251
Traceback (most recent call last):
  File "/nfs/users/stefano.carrazza/h.py", line 14, in <module>
    result = c(nshots=100)
  File "/nfs/tools/qibo/lib/python3.10/site-packages/qibo/models/circuit.py", line 1116, in __call__
    return self.execute(initial_state=initial_state, nshots=nshots)
  File "/nfs/tools/qibo/lib/python3.10/site-packages/qibo/models/circuit.py", line 1105, in execute
    transpiler = _Global.transpiler()
  File "/nfs/tools/qibo/lib/python3.10/site-packages/qibo/backends/__init__.py", line 126, in transpiler
    cls._transpiler = cls._default_transpiler()
  File "/nfs/tools/qibo/lib/python3.10/site-packages/qibo/backends/__init__.py", line 163, in _default_transpiler
    Unroller(NativeGates[natives]),
  File "/nfs/tools/qibo/lib/python3.10/site-packages/qibo/transpiler/unroller.py", line 26, in __getitem__
    return reduce(or_, [cls[key] for key in keys])  # pylint: disable=E1136
  File "/nfs/tools/qibo/lib/python3.10/site-packages/qibo/transpiler/unroller.py", line 26, in <listcomp>
    return reduce(or_, [cls[key] for key in keys])  # pylint: disable=E1136
  File "/nfs/tools/qibo/lib/python3.10/site-packages/qibo/transpiler/unroller.py", line 25, in __getitem__
    return super().__getitem__(keys)
  File "/usr/lib/python3.10/enum.py", line 440, in __getitem__
    return cls._member_map_[name]
KeyError: 'GPI'
```
@csookim could you please check?

stavros11, 2024-11-03:
I think the fix is in #1504.

alecandido, 2024-11-03:
> I think the fix is in [qiboteam/qibo#1504](https://github.com/qiboteam/qibo/pull/1504).

Indeed, I believe that was the main motivation behind qiboteam/qibo#1504

scarrazza, 2024-11-03:
We should do more testing before making releases to avoid these kinds of bugs.

csookim, 2024-11-04:
The issue is on the Qibo side: Qibo was supposed to raise an error when it received a gate that cannot be used as a native gate.

https://github.com/qiboteam/qibo/blob/22d51946add6739a3e970f32ee8cdd0bd70f5d2d/src/qibo/transpiler/unroller.py#L67-L68

https://github.com/qiboteam/qibo/blob/22d51946add6739a3e970f32ee8cdd0bd70f5d2d/tests/test_transpiler_unroller.py#L19-L21

I have merged https://github.com/qiboteam/qibo/pull/1504, and we should patch it to enable circuit execution. Do we need to wait for qibo `0.2.14`?

alecandido, 2024-11-04:
> I have merged [qiboteam/qibo#1504](https://github.com/qiboteam/qibo/pull/1504), and we should patch it to enable circuit execution. Do we need to wait for qibo `0.2.14`?

We need the `.wire_names` in any case, so we should complete https://github.com/qiboteam/qibo/pull/1500. No reason to anticipate the next release before that.

Current testing can be done with `master`. Releasing before that may lead to a similar problem with the next one (missing required features/not enough testing).

scarrazza, 2024-11-04:
On top of that, we should try to propose symultaneously a set of runcards (or sliced runcards) that works for at least the lines we are calibrating (D and B) for qw11q. At the moment, setting more than 11 qubits will generate an error message.

scarrazza, 2024-11-04:
Thanks @alecandido, indeed with 16 qubits it works. I have installed the latest qibo main in the cluster.

alecandido, 2024-11-04:
@csookim could you investigate and report about the mentioned error?

scarrazza, 2024-11-04:
I believe this is okay @alecandido. My example was using 17 qubits instead of 16. Using 16 works fine.

csookim, 2024-11-04:
> @csookim could you investigate and report about the mentioned error?

The `KeyError: 'GPI' error`? Yes, this occurred because there wasn't a specific rule to handle native gates between qibo and qibolab. https://github.com/qiboteam/qibo/pull/1504 resolved this issue. I am testing whether transpiler works well by creating `qw5q_dummy` and `qw11q_dummy`.

@scarrazza Could you give me permission to create a branch in `qibolab_platforms_qrc`? I would like to make a branch that includes the `qw5q_dummy` and `qw11q_dummy` models. This branch will not be merged; just for simulating the connection on the CPU.

scarrazza, 2024-11-04:
@csookim please go ahead, you should have write access now.

csookim, 2024-11-04:
Execution example:

https://colab.research.google.com/drive/1n-B9vPomUnyb1eKIjFoMo9N8cLd-kCWw?usp=sharing

alecandido, 2024-11-28:
@scarrazza did you have a chance to test whether this problem is still there? Hopefully, now that qiboteam/qibo#1504 has been merged, it should have been finally solved :)

scarrazza, 2024-11-28:
Yes, we can close.

closed 2024-11-28 by scarrazza