qiboteam/qibolab #1125: Import error of numpy when trying to run mixer calibration on 0.1 version of qiblab

Luca-Ben-Herrmann, 2024-12-27:
When trying to run mixer calibration (I copied and modified the mixer calibration script that middelware uses for the 11q chip ) for the quantum machine on the qw5q_platinum using 0.1 version of qibolab I get an error with the import of numpy (see attached file). I appears that numpy is trying to import platform which does not exist. 

 ```bash
File "/home/users/luca.herrmann/.envs/lbh_dev/lib/python3.10/site-packages/numpy/lib/utils.py", line 8, in <module>
    import platform
  File "/nfs/users/luca.herrmann/qibolab_platforms_qrc/qw5q_platinum/platform.py", line 3, in <module>
    from qibolab.channels import Channel, ChannelMap
```

[_qw5q_platinum_mixers.log](https://github.com/user-attachments/files/18258401/_qw5q_platinum_mixers.log)

stavros11, 2024-12-27:
Thank you for reporting this. I see from the attached logs that the mixer calibration script is executed from within the `qw5q_platinum` directory. This directory contains a `platform.py` file which I think is conflicting with an internal numpy import. Could you try moving the script to a different directory (one that does not contain any `platform.py`) and execute from there?

alecandido, 2024-12-29:
Indeed, `platform` is a [standard library module](https://docs.python.org/3/library/platform.html). Unfortunately, Python has the (arguably bad) habit of adding the current directory to its module path...

Luca-Ben-Herrmann, 2024-12-30:
Thank you for your help, indeed the issue was that platform is a standard library module. After moving the file to a different directory as well as the additional fix by Stavros (https://github.com/qiboteam/qibolab/pull/1127) I can now run the mixer calibration!

closed 2024-12-30 by Luca-Ben-Herrmann