qiboteam/qibolab #1135: Drive pulse for couplers in 0.1

jykhoo1987, 2025-01-22:
Is it possible to add drive pulse to couplers so that couplers have e.g. couplerdrive and couplerflux pulses? I am implementing a two-qubit gate pulse sequence that involves driving at a frequency different from any of the drive pulses for qubits and I am unable to do so within the current framework where each qubit is only allocated a single drive pulse and the other pulse channels are flux type (flux/couplerflux) that do not allow its frequency to be set from the runcard. For now I am adding an additional fake qubit just to make use of its allocated drive pulse for this purpose.

alecandido, 2025-01-22:
So, the actual answers are that:
1. couplers have no drive lines, that's why there has been no couplerdrive pulse
2. in 0.2 the couplerflux has been removed, and there is not even a flux pulse at all, since it is just a `Pulse` that is played on a DC line - if you want a `Pulse`, whatever, they all belong to the same class, and they are then played on different channels (modulated or not, and with the frequency attached to the channel - of which you may have many)

However, if you're still working on 0.1, the solution is simply to use the `frequency` attribute of the `Pulse` itself, so you should not need any new channel or different pulse class
https://github.com/qiboteam/qibolab/blob/bfb48cc2d9ec5e563110720889dc74acca6e604c/src/qibolab/pulses.py#L793-L797

jykhoo1987, 2025-01-22:
Could you clarify on the 0.1 solution? My scenario is that the coupler frequency pulse and qubit frequency pulse is implemented on hardware on two different channels, so that just with 2 qubits I do not have enough drive pulse channels to match the number of distinct hardware channels (3 of them, let's call them XY1, XY2 and XYC). Changing the frequency attribute doesn't switch the other details of the channel such as the port details; in addition my goal is to add this coupler frequency pulse as part of the calibrated CNOT gate on the runcard. I would like to write this coupler frequency pulse executed on XYC as part of the calibrated two qubit gate pulse sequence in the runcard. In this case I will need the pulse type to be 'qd', but without introducing an additional fake qubit, I have to choose either qubit 1 or 2 but that would assign XY1 or XY2 instead of XYC.

alecandido, 2025-01-22:
Just to summarize what has been said in the Qibo meeting: in 0.1 is mandatory to have a further qubit, because that's the way channels are organized, and there are a fixed amount per qubit.

In 0.2 you may use either a new qubit as well, or one of the spare (unlimited) channels of existing ones (under `drive_qudits`).