qiboteam/qibolab #1136: Add drivers and API for QRNG

scarrazza, 2025-01-22:
As discussed, we should consider including in Qibolab the control drivers, postprocessing routines and documentation for QRNG devices, starting from those manufactured by TII. In terms of features and API, we can try to replicate the [numpy random module](https://numpy.org/doc/stable/reference/random/index.html) with the addition of custom methods to tweak the device, e.g. entropy values.

alecandido, 2025-01-22:
I'd suggest making this a subpackage on its own.

Other than that:

> In terms of features and API, we can try to replicate the [numpy random module](https://numpy.org/doc/stable/reference/random/index.html) with the addition of custom methods to tweak the device, e.g. entropy values.

it should not be possible to actually reproduce the module, but only targeting a few functions.
Which is perfectly reasonable, but I would inspect them first, and decide ahead of time which is the target, since full API compatibility would be meaningless.

In particular, NumPy decided to even strength the accent on pRNG, by suggesting using the object-oriented API, such that the seed initialization and state evolution are explicit to the user.
If I understood correctly the scope of the project, no seed nor state will be stored on software, since that will be a property of the device (I guess you may have an initialization procedure, but it will be hardly comparable with a seed, since not having reproducible sequences will be a feature).

RoyStegeman, 2026-11-02:
@scarrazza can this be closed? It looks mostly implemented in #1138.

scarrazza, 2026-11-02:
Yes, absolutely, thanks.

closed 2026-11-02 by scarrazza