qiboteam/qibolab #1163: Qblox: Update of mixer phase and gain ratio parameters

jevillegasdatTII, 2025-02-28:
Hey! The current drivers (v0.1) update only the offsets (I and Q) of the mixer, which is useful to suppress the LO leakage, but do not update the gain ration and phase offset of the sequencers used, which allows to also suppress the image frequency. In QBlox this corresponds to the `mixer_corr_gain_ratio` and `mixer_corr_phase_offset_degree` parameters. And this applies to both QRM and QCM modules. An option to this in v0.1 for the QRM module is in https://github.com/qiboteam/qibolab/commit/7ca32a8689c0fdb01ee0796d71b78de7a497a286 , using the "mixer_readout_phi" and "mixer_readout_g" parameters in the platform specification.

e.g.
```yaml
 "characterization": {
        "single_qubit": {
            "0": {
                ...
                "mixer_drive_g": 1.0,
                "mixer_drive_phi": 0.0,
                "mixer_readout_g": 1.04,
                "mixer_readout_phi": 15.73
            },
      }
}
```

Can this be taken into consideration for the drivers in `v0.2`, and can we use as a current solution something like the commit referenced?

Some prior discussion about this is in #900.

alecandido, 2025-02-28:
Thanks @jevillegasdatTII for the issue. For sure, we can and will take it into account.

As you know very well, the v0.2 driver has still to lend in `main` or any release. However, I will certainly add this to the to-do list (partially, it was already there, as the first point of #1091, to which I will now add the reference to this issue).

However, since we want to reduce duplication across drivers, and standardize the access to common features, we'd like more to expose the options in a hardware-agnostic fashion, as long as it is feasible.
We already have (in v0.2) a configuration slot for mixers parameters. 
https://github.com/qiboteam/qibolab/blob/fb8a9b36a57c777e1bca89d381d50e812d3c5cb7/src/qibolab/_core/components/configs.py#L50-L69
It seems to me to be already compliant with what you're asking for. If you confirm that, I will label this issue as just related to Qblox, and proceed to implement it as soon as the rest of the driver works properly.

jevillegasdatTII, 2025-03-01:
Hey @alecandido, it does indeed look compliant. That said, the gain ratio and phase offset are normally adjusted on the waveform of the I and Q components, and can therefore be implemented in the **sequencers**, and usually this is better because this way two pulses using the same channel but targeting different frequencies can be individually calibrated, which you normally need to (especially for the phase offset). This is how QBLOX does it using ``` qblox_instruments.Module.sequencer0.mixer_corr_gain_ratio(#gain_ratio) ```. 

For drive lines that are not multiplexed the definition makes no difference, but for readout lines, multiplexed drives or drives for cross-resonance pulses (that all carry signals with different frequencies) they would be stuck to use all the same mixer calibration parameters as its currently implemented. Wouldn't they?

Adding this calibration as part of the gate makes sure that not only is unique for each pulse, but that it is implemented in the correct sequencer upon synthesis.

alecandido, 2025-03-01:
Hi @jevillegasdatTII, thanks for the detailed reply.

I'm actually familiar with the way Qblox implements this, and they can actually be changed in the sequencer (as you said), but also in real time, in addition to the configuration you are describing.
The picture is the following

<p align="center">
[image]
</p>

https://docs.qblox.com/en/main/cluster/q1_sequence_processor.html#awg-path

The configuration you're describing is represented as "Qcodes" in the diagram, while the real time ones are marked with a single letter (still in parentheses).

However, while changing them in real time may make it possible to apply different corrections for each pulse, I expect them to be useful mainly for sweeps. Instead, what you usually need is to adjust them per-sequencer.

In Qibolab v0.2 the frequency is associated to the channel. Consequently, in the driver I'm writing, each channel is associated to its unique sequencer, despite being connected to the same physical port.
So, whatever kind of multiplexing on ports (readouts, drives, and even cross-resonance frequencies, or drives for higher levels) are now associated to multiple sequencers by default, each with its own frequency.

Moreover, each RF channel is associated to its own mixer configuration
https://github.com/qiboteam/qibolab/blob/fb8a9b36a57c777e1bca89d381d50e812d3c5cb7/src/qibolab/_core/components/channels.py#L46-L54
(despite the lookup mechanism is such that here we're storing a single `str`, that's just an identifier to retrieve the `IqMixerConfig` object I posted above)

**In summary**, you are expected to be able to spell out the mixer calibration for each frequency involved.

If you believe it is not sufficient for your application, and you have further requests/advice, do not hesitate to let me know! (i.e. here below)

---

This scheme overall will have multiple limitations, but I expect them to come from the number of sequencers required whenever we will start playing with more sophisticated experiments (involving qudits and cross-resonances).
However, to avoid premature optimization, I did not take into account immediately the most general case, and we will consider whether it is possible to reuse some sequencers in case they will be needed (but the channels will need to be mutually exclusive, which in principle it is not guaranteed if they are multiplexed - so, it's a more complex problem overall, and we need to gather more details to conceive an effective design).

jevillegasdatTII, 2025-03-02:
Hey @alecandido . Yes it does indeed appear that this is the way to set it up. Being able to set calibration parameters for each frequency setting is exactly what we need. 🎉

jevillegasdatTII, 2025-03-17:
@alecandido do you think we can merge the current solution to 0.1?

alecandido, 2025-03-17:
> [@alecandido](https://github.com/alecandido?rgh-link-date=2025-03-17T13%3A33%3A50.000Z) do you think we can merge the current solution to 0.1?

Usual answer: nothing is impossible, it's just a matter of time. But, to a certain extent, the current solution is what makes 0.2 different from 0.1.

What we could do (or better could have done) is to port the 0.1 driver to 0.2, instead of rewriting from scratch.

Anyway, if the 0.2 driver will be actually confirmed to work, you could just start using that.
In case you have an existing calibration, we can (almost) automatically update it with a script which is in `qibolab_platforms_qrc` (https://github.com/qiboteam/qibolab_platforms_qrc/blob/0.1/convert.py).
In case you have your own routines for 0.1, which are not yet in 0.2, it should be fairly quick to upgrade. If you use circuits, there is (almost) nothing to be changed. If you use pulses directly, there are a few things, but not much (only the acquisition is affected by the switch from 0.1 to 0.2, since it's the only part in which enters Qibolab).