qiboteam/qibolab #1166: Unrolling and split batches

alecandido, 2025-03-05:
### Subdividing

Since `AveragingMode.SINGLESHOT` implies a `AcquisitionType.CYCLIC` mode, then shots are always executed as the outermost loop.
However, subdividing batches automatically in software loops is not bound to shots alone, since each sweeper loop is always required to be saved in independent bins, consuming acquisition memory as well.
Even these internal loops can be broken down in smaller chunks, but they would require a more sophisticated recombination - requiring multidimensional stacking, instead of stacking over the single outermost dimension.

In case this becomes useful, for any

### Merging

Unrolling is actually the opposite operation, merging multiple experiments, to leverage spare memory of all kinds (while for sub-batches only acquisition memory is relevant, since all the waveforms, weights, and instructions are already present for the individual bin).

## All at once

The full problem thus consists in optimizing
- the length of sequence batches
- the number of internal software-looped subdivisions, within a single sequence

subject to the constraints of acquisition, waveform, weight, and instructions memory.

This problem has to be solved for all channels simultaneously, which mostly boils down to picking the most stringent constraints (not completely true, since sometimes it may be possible to trade more subdivisions for multiple simultaneous sequences - but we can simplify the problem preventing unrolling in presence of split batches).

Though parameterized by the memory constraints (possibly, per-channel constraints), the problem is fully hardware-agnostic, thus we can solve it outside the individual drivers, just building joint sequences and split batches around the `Controller.play()` invocations.

RoyStegeman, 2026-09-09:
@alecandido I would like to close this one for the same reasons as stated in https://github.com/qiboteam/qibolab/issues/698#issuecomment-5601694972. However, in this case I'll discuss first because the issue was opened more recently and it states the following

> Though parameterized by the memory constraints (possibly, per-channel constraints), the problem is fully hardware-agnostic, thus we can solve it outside the individual drivers, just building joint sequences and split batches around the `Controller.play()` invocations.

I believe that instead we want to do the unrolling+batching in the driver since the type of constraint may be different for different electronics. We simply hand over a list of pulse sequences to the driver, and the driver decides how to batch them. If a driver does not yet support batching that also doesn't break anything and it can just run each pulse sequence individually.

alecandido, 2026-09-09:
Agreed. This was opened at the time I've been able to run my first sequence with Qblox, and I was preparing a list of leftover. It was highly inspired by #698, and written before any attempt to actually implement unrolling.

By now, unrolling has been implemented for Qblox, and it is much more advanced than what I envisioned here. It is not completely done, and we have pending improvements (e.g. regarding the instructions memory, which is currently batched on a fit based on the RB). But, still, there is no point in following this any longer.

closed 2026-09-09 by alecandido