qiboteam/qibolab #1174: Update IIR filter implementation for OPX1000

Edoardo-Pedicillo, 2025-03-13:
According to the QUA documentation, the API for applying IIR filters is different between OPX1000 and the others (https://docs.quantum-machines.co/1.2.2/docs/Guides/output_filter/?h=taps#using-the-filters). 
Instead of the IIR coefficients, the exponentials and the amplitudes are required. Qibolab does not take this difference into account and uses the "feedback" key for both cases. 
This key seems supported in OPX1000 since no error is raised (even though we observed some strange behaviors in the number of IIR we can use), but it is not documented. I propose the cryoscope to return the FIRs, and the exponentials for the IIR and Qibolab will translate the IIR according to the control electronics.

andrea-pasquale, 2025-03-13:
Yes, in Qibocal it should be feasible to make the change.
The only problem is that we will also need soon the corresponding PR in qibolab to load the filters in the correct way, otherwise we cannot use the cryoscope (unless we leave to the user the platform update).

alecandido, 2025-03-13:
> This key seems supported in OPX1000 since no error is raised ~(even though we observed some strange behaviors in the number of IIR we can use)~, but it is not documented. I propose the cryoscope to return the FIRs, and the exponentials for the IIR and Qibolab will translate the IIR according to the control electronics. [[image]@andrea-pasquale](https://github.com/andrea-pasquale?rgh-link-date=2025-03-13T06%3A54%3A24.000Z), do you think we can change it in [qiboteam/qibocal#974](https://github.com/qiboteam/qibocal/pull/974) ?

The fact that is not raising does not guarantee that it is actually consumed by the electronics. I believe they are only validating a unique JSON schema, which is the union of OPX+ and OPX1000, but they are not raising if there are keys that are only consumed by the other tool.

In particular, the OPX1000 controllers are below the `fem` key (since OPX1000 introduced modules in QM, while OPX+ was exposed as monolithic),

The schema itself is published as part of the docs, and it documents the keys in there
<details><summary><a href="https://docs.quantum-machines.co/latest/assets/qua_config.html#/paths/~1/get">Schema documentation</a></summary>
<p>

[image]

</p>
</details> 

At least, the keys are confirmed to be the exact same.

alecandido, 2025-03-13:
> I propose the cryoscope to return the FIRs, and the exponentials for the IIR and Qibolab will translate the IIR according to the control electronics. [...]

> The only problem is that we will also need soon the corresponding PR in qibolab to load the filters in the correct way, otherwise we cannot use the cryoscope (unless we leave to the user the platform update).

I agree with the proposal, and I agree to patch the handling of the IIR in Qibolab, just moving the piece of code currently computing IIR from exponentials from Qibocal, and adopting it when needed.

Notice that

[image]

is also listed below the keys used by OPX+, so there is a chance that exponentials are supported even on this device (or they just reused the subschema for filters in both instances, though only OPX1000 is capable of using them).

stavros11, 2025-03-13:
> Notice that
> 
> <img alt="Image" width="407" src="https://private-user-images.githubusercontent.com/10830876/422272950-255004e7-f538-40df-ab67-a0b7aa57827c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDE4NjQ1ODEsIm5iZiI6MTc0MTg2NDI4MSwicGF0aCI6Ii8xMDgzMDg3Ni80MjIyNzI5NTAtMjU1MDA0ZTctZjUzOC00MGRmLWFiNjctYTBiN2FhNTc4MjdjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAzMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMzEzVDExMTEyMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTY2ZWZjNGMxOTJiOGE3OTc2ZmQ3YmQxZTA2YmYyODAyNDMwNDJkZGMzZWVlNTNmNDRiMjU0ZWUxYmJlYWQ2Y2QmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.p2geTtS9fy6PR5MGDp_P_EnNbHyGH2zYxtKETSBgelQ">
> is also listed below the keys used by OPX+, so there is a chance that exponentials are supported even on this device (or they just reused the subschema for filters in both instances, though only OPX1000 is capable of using them).

Looking at this, it seems that neither of our clusters will support the `exponential` option as they are now, since we are running the following QOPs:
* OPX+: QOP 2.4.2
* OPX1000: QOP 3.2.3

so both are <3.3. I guess 3.3 came out after we last upgraded. It should be possible to upgrade our OPX1000 to 3.3 (with all the usual risks and potential incompatibilities) and change the driver to accept `exponential` instead of `feedback` for the OPX1000 only. Regarding the OPX+, if I understand correctly the QOP versioning conventions, it will always get QOP2 (not 3) so it will still not support the `exponential`. However, this is just my assumption (I am not sure) and they could even release `exponential` support for QOP2 at some point.

What I don't really understand from the documentation is if the `feedback` option is dropped after `exponential` is supported, or they still coexist (and both work properly). If the first is true, then we will need to update the driver together with the upgrade to QOP3.3, otherwise we could still get away with using `feedback`.

In summary, if there is strong motivation to use `exponential` with OPX1000, I would first try to run a quick QUA program with it with our current QOP. If the schema documentation it correct, this should fail. Then I would upgrade the QOP and the driver at the same time.

alecandido, 2025-03-13:
Thanks @stavros11. I was not aware of the QOP versions on the lab clusters, but if QOP 3.3 is recent enough, I would say that it is not that urgent to upgrade just for this reason (with the usual potential consequences...).

However, I believe that the support for `exponential` will be just implemented in the very same way it is currently implemented in Qibocal. In order to apply it in real time, they will need taps (having one FPU for each channel computing exponential suppressions for each sample in real time is not feasible - at that point, you don't do it in real-time, and just use `lfilter`).

So, if it was not supporting `exponential`, even the OPX1000 should have supported taps, and it should be fine as it is.
If there are observed issues, we should understand where they are coming from, possibly getting in touch with the vendor (if the docs are not sufficient, and obtaining a clearer answer is feasible).

RoyStegeman, 2026-11-02:
@alecandido is this closed with #1548?

alecandido, 2026-11-02:
Yes, it should. But it is yet untested.