qiboteam/qibolab #1190: Are these platforms open to public?

Wangwei-Lan, 2025-03-29:
Hi, to whom it may be concerned, 
I am actually very interested in the qibo library developed by your team. However, I don't have any access to a real QPU platform, nor did I found a virtual QPU that is supported by qibo. So I was wondering are the platforms here available to general public? Thanks very much. 

Yours sincerely
Wangwei

alecandido, 2025-03-29:
@Wangwei-Lan, given the present scope of the issue, I moved it to the Qibolab repo (where any development would be expected to happen), and adjusted its title.

Let me know if you disagree with any of the changes.

Wangwei-Lan, 2025-03-30:
@alecandido Thanks again for your very detailed instructions and that's very helpful. Please feel free to make any modifications of the original post to be in line with your development better.