qiboteam/qibolab #1192: Align numpy and python versions

scarrazza, 2025-04-03:
At the moment numpy and python versions in the project toml are different from qibo.
Could we align both versions to match those in qibo?

alecandido, 2025-04-14:
In principle, there is no problem for NumPy. I'll try soon.

Concerning Python 3.13 we have a problem instead. It is not supported by `qm-qua`. Despite being optional, this will prevent locking.

Actually, I foresaw a similar problem some time ago, and I started conceiving a plan to reduce the present exposure to transitive dependencies.
The best way would be to deploy drivers in separate packages, instead of bundling them within Qibolab itself, and keeping their dependencies as extras. No need for a new repo for each of them, as we can keep these packages in a subfolder. I'll generalize workflows to cater for this structure as well (I just need to specify a subfolder, and to allow for publication failures, since we do not have to update all drivers at each repo tag/release). But, of course, even different repos remain an option.

I would have liked to stabilize some part of the internal interface as public, in order to support external drivers as the first step. However, given the need, we could even externalize first existing drivers, to reduce the dependency complexity (though interfaced with a yet unstable mechanism).

alecandido, 2025-04-16:
NumPy is blocked by the same problem for py3.13. We will then solve this issue in a single step.

```py
❯ : poetry lock
Resolving dependencies... (1.6s)

Because qm-qua (1.2.2) depends on numpy (>=1.17.0,<2)
 and qibolab depends on numpy (^2.0.0), qm-qua is forbidden.
So, because qibolab depends on qm-qua (==1.2.2), version solving failed.
```

stavros11, 2025-04-19:
Thanks for checking @alecandido. If qm-qua requires numpy<2 and qibo requires numpy>=2.0, won't this be an issue for the cluster installation, if we want to be able to run Qibo circuits on QM? Regardless of how we distribute qibolab drivers and dependencies, if we want to use QM, we will need the "qua subpackage" and this will be incompatible with qibo...

alecandido, 2025-04-20:
Yes, formally it is an issue. But only formally, I believe.

In practice, NumPy 2 broke the compatibility with previous versions, but not completely. Most of NumPy functions and classes are still working in the exact same way in NumPy 1 and 2. Most likely, both current Qibo and qm-qua can actually work with both versions (we could even test this in Qibo, and lift the restriction on NumPy, accepting both v1 and v2).

Since this is certainly a formal problem, it will result in a problem for locking, if Qibo and qm-qua should be locked in the exact same lockfile.
In practice:
- if we split qm-qua from Qibolab, we can separately install it (`pip` is not complaining about inconsistent versions, it will just make `qm-qua` dependencies consistent - but we can even revert that)
- we can then decide which version of NumPy we like the most
- we could lift the NumPy requirement in newer Qibo versions (I'm not completely sure it's possible, but we can check)
- we can keep working with an older Qibo version (newer versions should introduce features, but I do not see much of what we recently introduced as mandatory)

Furthermore, lockfiles are a useful tool for reproducibility and development, but they are by no means required on the user side, or for environment management (and we could also decide to manage only part of the environment using them). If there is any practical conflict issue, we could handle it manually.

(The present mess is mostly caused by rushing for new-version adoption. In general, we should try to impose as mild as possible limitations on our dependencies, and let them be restricted by other people. However, we transitioned to NumPy 2 because of downstream pressure. And now we're possibly about to generate downstream pressure on qm-qua as well, because most people require compatibility with a single major. Let's try to be more lenient in the future, and slow down transitions)

BrunoLiegiBastonLiegi, 2025-05-20:
For the record, I am trying to test qibolab (through the emulator) with qiboml here qiboteam/qiboml#94, which is working fine locally. However, I cannot include qibolab into the test dependencies due to the numpy version mismatch. For the moment, I guess I will have to only run the test locally and not in the CI.

alecandido, 2025-05-20:
> For the record, I am trying to test qibolab (through the emulator) with qiboml here [qiboteam/qiboml#94](https://github.com/qiboteam/qiboml/pull/94), which is working fine locally. However, I cannot [...]

Solution in progress :)

stavros11, 2025-05-20:
> For the record, I am trying to test qibolab (through the emulator) with qiboml here [qiboteam/qiboml#94](https://github.com/qiboteam/qiboml/pull/94), which is working fine locally. However, I cannot [...]

@BrunoLiegiBastonLiegi if you have time, try to point the CI to the branch from #1221 to see if it works. This should be compatible with numpy>2.0 and the latest qibo.