qiboteam/qibolab #1199: Inquiry on how to connect RFSoC equipment

gksmfdl1707, 2025-04-10:
Since v0.2.x, is qibolab.instruments.rfsoc not supported?

Is there a way to interface with rfsoc equipment?

If you have an example, please let me know.

alecandido, 2025-04-11:
> Since v0.2.x, is qibolab.instruments.rfsoc not supported?

Not yet, I'm working on #1175 exactly to restore support for it. Hopefully, it will be back in ~1 month.

> Is there a way to interface with rfsoc equipment?

For the time being, we're still maintaining https://github.com/qiboteam/qibolab/tree/0.1 for bug fixes. If you need to use a QICK board with Qibolab right now, we advise the `0.1` branch usage (there are corresponding branches even in Qibocal and in https://github.com/qiboteam/qibolab_platforms_qrc/, same name).

gksmfdl1707, 2025-04-14:
> Not yet, I'm working on https://github.com/qiboteam/qibolab/pull/1175 exactly to restore support for it. Hopefully, it will be back in ~1 month.

I checked the source and it seems to be using qick.asm_v1. Does that mean that the version using asm_v2 is also supported?

zzwon-zzu, 2025-08-01:
Hi, how is it going on qibolab.instruments.rfsoc on qibolab v0.2.x?

alecandido, 2025-08-01:
> Hi, how is it going on qibolab.instruments.rfsoc on qibolab v0.2.x?

On hold until now, since we had to complete #1088 support. However, that's (more or less) done, so we'll resume #1175 soon™️

closed 2026-04-01 by alecandido