qiboteam/qibolab #1276: QM doesn't allow readout with acquisition and probe with different duration

andrea-pasquale, 2025-08-25:
As per title, I was trying to run using a different duration for the probe and the acquisition. However it seems to be not supported by the QM driver
https://github.com/qiboteam/qibolab/blob/a3754730532ba7e26917143ecb16a418f4f9e211/src/qibolab/_core/instruments/qm/controller.py#L488-L491
Is there a motivation behind this? Just be uncommenting the line I was able to get results that were making sense.

alecandido, 2025-08-26:
> I am not sure how I would do that in QM. Given that it is actually useful in practice, it is likely that they provide a way, so we could have a look.

Yeah. I love thinking about the options, and how I would do something. And usually refrain to just say "it's not possible", unless it is truly impossible for some reason.

However, I agree that the best answer most likely is "it is impractical". And wait for QM to provide such a feature, denying the option until then.

> Coincidentally, I was looking at the same issue for qblox yesterday (see [#1278](https://github.com/qiboteam/qibolab/pull/1278)). There it is actually possible to use different durations, since it has separate `play` and `acquire` instructions.

I was following the same bundle logic, since analyzing it as a bundle was convenient to allocate I/O sequencers.
However, it is true that it will be implemented as two separate instructions anyhow, so you have enough freedom to play with the durations. I will review the PR as soon as possible, thanks for it :)

andrea-pasquale, 2025-08-26:
Thanks for the explanation @alecandido @stavros11 
I was wondering this because I remember from previous discussions that we would like to do this at some point.
Let me know if you would like to keep the issue open as a reminder if not I can close it.

RoyStegeman, 2026-11-02:
Implemented in #1324: the QM driver now pads the probe with zeros up to the acquisition duration, so readouts with an acquisition longer than the probe are supported.

closed 2026-11-02 by RoyStegeman